OrcaSlicer/OrcaSlicer #834: Gap between outer wall and brim

Chodec80, 2023-04-23:
**Describe the bug**
A gap between outer wall and brim in case, that Elephant foot compensation is active. 

**To Reproduce**
Set Elephant foot compensation and activate brim. Set Brim-object gap to zero.

**Expected behavior**
No gap between outer wall and brim. Like Superslicer.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS 13.3.1
 - Orca: 1.6.2 Beta

tito, 2023-07-05:
Got the same issue, brim gap to 0, while the brim is perfectly disconnected of the whole object after printing.
[image]

LowRunneR, 2023-07-28:
The issue is also in 1.6.3 Official.
Is a fix possible?
please :-)

closed 2023-11-20 by github-actions[bot]

b3n3d1k7, 2024-05-19:
The issue is still there in 2.0.0