OrcaSlicer/OrcaSlicer #836: Add the ability to create custom build plates with their own properties

CitizenRacer, 2023-04-23:
**Is your feature request related to a problem? Please describe.**
I would like to be able to create new build plate types, for example, I made a G10 build plate for my Bambu X1C and I'd like to be able to configure it differently from the high temp plate.

**Describe the solution you'd like**
I would like to see the ability to create a new build plate and on create make it behave like one of the other plates unless new defaults are set. For example, I'd like to define a "G10 build plate" which behaves like a high temp build plate unless I set new values (the value I'd like to set is increase bed temp by 5º)

**Describe alternatives you've considered**
I can change the settings for the high temp plate every time I use it and then remember to change it back when I'm using a traditional high temp plate.

**Additional context**
I imagine this would be useful for others who choose to use glass build plates, or people with different printers where the defaults don't line up quite right.

TexZeTech, 2023-04-24:
I too would like to see this as I have a [buildtak](https://www.buildtak.com/) surface that actually needs to have a different z offset as to not destroy its surface and if I put something else on like G10 I suspect it would be vastly different offset.

julie777, 2023-06-02:
I cannot remember which issue number, but I requested that in addition to global settings having the option to select printer, filament and process profiles, that we should provide those same choices on a per plate basis. My thought is for a complex project with many plates, that objects will be separated onto separate plates based on
- core parts and separate groups of optional parts
- type of filament needed to print them
- needing supports
- common process profile
- etc.

There are many cases in my projects when I have a plate of objects and it uses a different set of profiles than the other plates. Not being able to apply the settings for that plate using profiles is a huge problem.

jphoke, 2023-06-12:
I would love to see this for a multitude of plates and use cases. Yes you can do it as different printer settings, but that's not ideal IMO

JeffreyBrew, 2023-10-01:
I would like to be able to add custom build plates too.  I have a smooth PEI plate I would like to add.  Selecting the Textured PEI plate then creating separate process profiles is a pain.

Chinstrap777, 2023-11-15:
I would like to switch between my K1 Max Textured PEI sheet and Smooth Sheet. Is there a manual way to build it?

Rodamyot, 2023-12-04:
Me too I'd like to be able to create custom Bed Type. Textured bed need some adjustment on the Z height for good bed adhesion.
It would be logical to put that in the accessory section of the printer Basic information. And some kind of way to associate Z offset to it and maybe some temperature offset as well.

TexZeTech, 2023-12-04:
> Me too I'd like to be able to create custom Bed Type. Textured bed need some adjustment on the Z height for good bed adhesion. It would be logical to put that in the accessory section of the printer [...]

I think that could be handled with just gcode or passing "bed type" to the printer if its klipper-like.

I really would like to have that option enabled though.

Rodamyot, 2023-12-04:
@TexZeTech a lot of thing can be done with G-code, well almost everything. But there's a menu (Bed Type) in the Gui that seems to be useless! Unless you have a bambulab. So let's make work!

zchristopoulos, 2024-02-19:
+1, I would love to see this.

MrGrendel27, 2024-03-24:
Just started working with Orca slicer and I too would like to see this option added. I like to use smooth PEI, PEO, and PET. I would like to rename each bed type they have listed to my type of bed I'm using.

vgdh, 2024-03-24:
related https://github.com/SoftFever/OrcaSlicer/issues/4655

If you can choose custom names, you can use them in Klipper macros.
The structure may be inherit like.
    Glass (it would appear at all profiles as a plate name)
    |---- favorite glass (doesn't apper at profiles but this will be sent to klipper as placeholder)
    |---- rough glass (doesn't apper at profiles but this will be sent to klipper as placeholder)
    PEI (it would appear at all profiles as a plate name)
    |---- (name isn't chosen so 'PEI' will be sent to klipper as placeholder)

Chinstrap777, 2024-03-25:
> Just started working with Orca slicer and I too would like to see this option added. I like to use smooth PEI, PEO, and PET. I would like to rename each bed type they have listed to my type of bed I'm using.

Looks like they just released the option with v2.0! Love it

TexZeTech, 2024-03-31:
> > Just started working with Orca slicer and I too would like to see this option added. I like to use smooth PEI, PEO, and PET. I would like to rename each bed type they have listed to my type of bed I'm using.
> 
> Looks like they just released the option with v2.0! Love it

I had to go digging & so others don't here is a link to said info.

https://github.com/SoftFever/OrcaSlicer/wiki/bed-types

This is definitely a step in the right direction.

Chinstrap777, 2024-03-31:
Yes! Thank you Softfever!

Hawkeye666, 2024-04-26:
> I had to go digging & so others don't here is a link to said info.
>
> https://github.com/SoftFever/OrcaSlicer/wiki/bed-types
>
> This is definitely a step in the right direction.

I do not see this available for my X1-C in Orca 2.0.0. But if I add a Creality or similar I have it for that printer. What gives. How do I enable this feature?

kylek29, 2024-06-03:
> I do not see this available for my X1-C in Orca 2.0.0. But if I add a Creality or similar I have it for that printer. What gives. How do I enable this feature?

Same here for my X1-C. Maybe they disabled it because the BambuLab uses the barcodes?

A work-around for that would be to make it so that the BambuLab users "inherit" from one of the existing BL presets and the new plate acts as an alias to one of those.

E.g. The user creates a new plate called "Plate - G10 Smooth" and sets it to be an alias for Heat Plate.

JekleFPV, 2024-08-03:
> I cannot remember which issue number, but I requested that in addition to global settings having the option to select printer, filament and process profiles, that we should provide those same [...]

Very much this!

I try to maintain just 1 'Project' file per project, on a single-material printer, even though I often use different filaments within the same project. So I sort them in to plates that each only use one filament type.
It would be nice to have Orca to arrange the objects into separate plates based on the set filament, but that's not even my main concern tbh.

I run a filament manager (Spoolman) on my printer, but as it is now, it is still a hassle every single time. I have to create separate files for every filament type within my project.
So the main thing for me would be to only include the filament in the G-code on a per-build-plate basis, instead of what happens now where it includes all the filaments within the entire project. Which in turn would make it possible to have only 1 'Project' file per project.

Even if there's no auto-arrange based on filament, per-plate-filament settings/G-code inclusion combined with this feature request would be absolutely golden for anyone running Klipper/Spoolman:
https://github.com/SoftFever/OrcaSlicer/issues/2955

scolgan0309, 2024-09-17:
I would love to be able to create a custom build plate profile for my bambu x1c.

AJolly, 2024-10-20:
Same, esp as a I have a number of different plates that really need a different z offset

Emperornero, 2024-11-15:
We need this feature added for Bambu printers with the recent release of Cryogrip, we should be able to set up settings for these plates, to allow running them with no/low bed heater without modifying other plate profiles.

bzzzbzzzzz, 2024-12-08:
There are really a lot of available build plates meanwhile. Just to list a few, just for Bambu Printers:
- BIQU Panda CryoGrip Pro Frostbite
- BIQU Panda CryoGrip Pro Glacier
- Darkmoon 3D Satin
- Darkmoon 3D ICE
- Darkmoon 3D Lux
- Darkmoon 3D G10 Hobby
- Darkmoon 3D G11 Pro
- Lightyear div.
- Sliceworx Plate
- VPSData G10
- VPSData G11 0.2 mm
- VPSData G11 1.5 mm
- Wham Bam (div.) PEX Plate(s)
- div. PEO Plates
- div. PET Plates
- div. PEY Plates

I'm sure one could use the same settings for some of those plates, for some or all filament types/profiles, but I think it'd be great if the user could decide,
instead of manually setting bed temperature,
or remembering that they abuse the Cool plate bed temperature setting for G11.

Having the ability to add custom named build plates would keep the build plate list clear/overseeable and would avoid to have to wait to use new build plate products, even if the main focus was only BambuLab products.

flashburn2012, 2024-12-08:
Bambu also just released a new build plate, Cool Plate SuperTack. We REALLY could use this ability.

I actually managed to scrape off the coating of one of my Cryogrips because I used the Smooth plate setting instead of Textured. This would be easier to avoid if we had the ability to add and name our own build plates in Orca.

mytechguyri, 2024-12-21:
Surely there should be some traction on this by now?   With all the different build surfaces coming out, this is a needed enhancement.

rashuge, 2024-12-27:
I would like to see this option available for the Bambu P1 series at the very least.  There is nothing that should be stopping this functionality as it is very similar in function to other printers without lidar.  It would be great to see this multiple bed type support available.  I have the issue where I bought BQ build plates and they simply don't work when I forget the temp change.  If there was a bed type, I wouldn't have this issue.  The Z offset option would be great too but there is always the option for G-Code and support the "build plate type" and having a Z offset.

ImNoahDev, 2024-12-29:
+1. Would be nice if we could make plates for Bambu and pick an arducode to associate with the plate but still have different settings.

TheKurgan, 2025-01-13:
Someone tell me how this isn't a thing yet. With all of the different types of plates out there, we are limited to just a few.

ZerkerEOD, 2025-02-23:
I'm in the same boat, would be nice to create the plate based on the others and set default temperatures for the new plate that can be over written by the filament for fine tuning. Also an option to prevent certain filament would be great also so that you don't accidentally print a filament that shouldn't go with that plate.

thx1200, 2025-03-06:
Glad to see so many others with the same request!  I really want this too.

Culain, 2025-03-28:
+1
I got the info that i should increase the bed temperature for print plates with an effect on them (3d effect like the galaxy plate from bambu). But i can't seem to modify any of the print plates values. So i would just add a new plate.
Bonus if you can set the first layer variables (speed and print temp) as well.

SoftFever, 2025-03-29:
> [@SoftFever](https://github.com/SoftFever) Please tell us how this isn't a thing yet. With all of the different types of plates out there, we are limited to just 5 (3 of which are very rarely used) I mean, I just started with Bambu printers, and I have 3 different build plates already... the Textured PEI it comes with, a Whambam PEX plate, and a BIQU CryoGrip Pro Glacier

Feel free to submit your code then

TheKurgan, 2025-03-29:
Hah! awesome, 2 1/2 months later I get a sarcastic response.
I am very sure that if the author and contributors are able to code what is already there, then they are more than capable of modifying this existing feature to accommodate this seemingly popular request, that was first submitted almost 2 years ago.
I do not posses the knowledge to actually do the coding, but does that mean I can't participate in this conversation, and ask questions?

thx1200, 2025-03-30:
FWIW, I did look at the code and it looks like it's not super complex but also not trivial.  Each build plate is a "hard coded" to a value, so it's not like this is simply a matter of adding some items to a list.  I'm sure this was a design decision that goes waaaaay back to the original source code, back where there just weren't really many build plates at all.

I think a reasonable solution that doesn't require recoding the whole thing would be to add another 5 (or so) "User" plates, also hard coded, and then give us a means to maybe give them a new name. 

I should point out that since I'm not a developer on this project and also just a hobbiest like most users, I may have missed something really big in the code and freely admit it may be more complex than it looks.  I also have not ever tried even building this project and have limited time to do that, but I might take a stab at it in the future, who knows.  But I am super hoping somebody else more experienced working on this project takes it upon themselves.  :-)

SoftFever, 2025-03-30:
> Hah! awesome, 2 1/2 months later I get a sarcastic response. I am very sure that if the author and contributors are able to code what is already there, then they are more than capable of modifying this existing feature to accommodate this seemingly popular request, that was first submitted almost 2 years ago. I do not posses the knowledge to actually do the coding, but does that mean I can't participate in this conversation, and ask questions without getting a 7 word sarcastic response?

No one is discouraging you from participating in the conversation. But when you ask in this manner, what else do you expect? Seriously, what do you expect me to reply when you ask in this way? 

> @SoftFever Please tell us how this isn't a thing yet.

SoftFever, 2025-03-30:
No plan to work on it

closed 2025-03-30 by SoftFever

Jeffrey-Brew, 2025-03-30:
@TheKurgan,

ZerkerEOD, 2025-03-30:
> > Hah! awesome, 2 1/2 months later I get a sarcastic response. I am very sure that if the author and contributors are able to code what is already there, then they are more than capable of modifying this existing feature to accommodate this seemingly popular request, that was first submitted almost 2 years ago. I do not posses the knowledge to actually do the coding, but does that mean I can't participate in this conversation, and ask questions without getting a 7 word sarcastic response?
> 
> No one is discouraging you from participating in the conversation. But when you ask in this way, what else do you expect?
> 
> > [@SoftFever](https://github.com/SoftFever) Please tell us how this isn't a thing yet.
> 
> P.S. I didn’t intend to be sarcastic in my post. Perhaps it’s not the most pleasant response, but I didn’t mean any harm or sarcasm.

I took it as pretty sarcastic as well. I see a lot of developers that contribute to open-source projects get an elitist mindset and like to tell people to basically do it themselves, knowing that more than likely most the people using their projects are not in fact programmer's. 

On a side note, I understand you marked it as not planned, but why? This feature would help not only help the community, but yourself as well if you still use Bamboo as a printer. I feel it should still be planned, even if it doesn't take a high priority.

thx1200, 2025-03-30:
Anybody else in this thread a dev?  I might try to implement something like I mentioned above, but I'm a bit intimidated by this project's size.  If a couple of us knuckle down we might be able to put together a PR that the dev would consider integrating.

johnny-bit, 2025-03-30:
> I see a lot of developers that contribute to open-source projects get an elitist mindset

Please note that contributing to open-source projects is mostly done as a hobby or very part-time. Encouraging people to contribute is a way to get out of problem of finite time and interests of a contributor/maintainer. For people who aren't developers (or don't have time to invest in the project) there's a possibility of offering a bounty for a problem/feature that might entice devs with time. If you can't offer your time or money to the project and still want to have the feature made (or issue fixed) the best way is to open a feature request and wait for somebody to pick the idea up.

> I might try to implement something like I mentioned above, but I'm a bit intimidated by this project's size.

Good on you for willing to try it! Don't be intimidated! Try checking out how build plates were done, then go from there. Maybe join discord and chat with devs there?

ZerkerEOD, 2025-03-30:
> Please note that contributing to open-source projects is mostly done as a hobby or very part-time. Encouraging people to contribute is a way to get out of problem of finite time and interests of a [...]

I understand, but it doesn't come off as encouraging them to help contribute. It looked more like a challenge and an assumption that he was talking to someone that couldn't actually create the code and PR. Hence my reference to elitist attitude.

TheKurgan, 2025-03-30:
@SoftFever  My question was not intended to be taken with any sort of rude or negative connotation, I was genuinely asking how it wasn't a thing yet. 
With the plethora of different build plates out there, with all of their different textures, materials and required temperature settings, I thought there would be the ability to set up custom plates, or at least have several more built-in plates that we could customize.
Basically telling me to do it myself came off as a little rude to me... especially because I LOVE OrcaSlicer, and have not used any other slicer since my Ender 3 Pro was brand new, and I used Cura.

SoftFever, 2025-04-09:
> [@SoftFever](https://github.com/SoftFever) My question was not intended to be taken with any sort of rude or negative connotation, I was genuinely asking how it wasn't a thing yet. With the plethora [...]

I must have brought bad energy from another thread. I didn't mean to upset you. 
Thank you for your understanding.

LupusTheCanine, 2025-06-14:
@SoftFever Hi, why the issue is marked as "completed" when it isn't. It is misleading.

LupusTheCanine, 2025-06-16:
> > why the issue is marked as "completed" when it isn't
> 
> The issue is "closed" and as to why it's clear from reading the thread. I'd encourage you to open a new feature request and ideally contribute to the project. Doesn't have to be code contribution, but maybe set a bounty for this feature? Also, please remember human on the other side.

The issue is closed as completed. It should be closed as duplicate (if it indeed is), not planned (if that is the case) or not closed.

Recognizing incorrectly closed issue requires reading through a large part of the discussion.

LupusTheCanine, 2026-01-28:
@rob-4x4 only in a custom build, by modifying code, but it will clash with default profiles.

@SoftFever this should be closed as duplicate (if it indeed is), not planned (if that is the case) or not closed.

impeeza, 2026-02-22:
I do not know why this is closed,  by example my Creality Hi comes from factory with a build plate «Epoxy Resin» type and needs some type of different settings, and is a pain in the harse to get it working perfect on Orca Slicer.

johnny-bit, 2026-02-23:
This request was closed after unnecessary hostility. I strongly suggest opening a new fresh one with positive attitude, clearly defined goals and maybe a bounty ;)

So far I've posted 4 bounties and I'd recommend posting one, very good experience and short time to get the feature one would like to see :)