OrcaSlicer/OrcaSlicer #837: Acceleration stays stuck at 500mm/s after first layer/top surfaces

harriettubgirl, 2023-04-23:
I'm not sure if anyone else is having this same issue but my ender 3 v2 klipper printer changes the accel speeds for top surface and first layer to 500mm/s. After those layer types, it's supposed to go back to 3000mm/s. It stays stuck at 500mm/s until I hit the little refresh/undo button. Picture after I refreshed it back to 3000mm/s: https://i.imgur.com/kQAZZwq.png 

I have all the settings as I thought they should be: https://i.imgur.com/QgeeZVR.png
Maximum accels:
https://i.imgur.com/dCB7cMC.png

Here's the g-code for the file if it's helpful at all: https://drive.google.com/file/d/1_GPqIKSp_KG7mZtvxeZztLQe-1lU_p8F/view?usp=share_link

If anyone is able to help out I'd really appreciate it! Thanks

SoftFever, 2023-04-26:
@harriettubgirl 
You have set `maximum acceleration for extruding` to 500 which is the cause. Usually, we can set it to the same value as `Maximum acceleration X`.
Also in the future, you can paste images here directly, which will be much easier

[image]

harriettubgirl, 2023-04-26:
Thank you! I don't know why I read that as maximum acceleration for the extruder haha. Appreciate all your hard work!

closed 2023-04-26 by harriettubgirl

1-ashraful-islam, 2024-02-08:
Maybe a tooltip can be added to this setting or change the description - After going through many different trials, online resources - this is what solved my slow print speeds! Wouldn't have guessed..