OrcaSlicer/OrcaSlicer #840: Thumbnails not working in Mainsail

jonathanalevi, 2023-04-24:
**Describe the bug**
For some reason, when I switched to OrcaSlicer, thumbnails stopped showing in Mainsail

**3mf File for This Bug**
Uploading 
[LDO_Main_Plate.zip](https://github.com/SoftFever/OrcaSlicer/files/11309079/LDO_Main_Plate.zip)

**To Reproduce**
Configure thumbnails, try to send to mainsail

**Expected behavior**
Thumbnail shows

**Screenshots**
[image]
**Printer model**

**Desktop (please complete the following information):**
 - OS: macOS but it's not relevant
 - Version [e.g. 22]

frankt2012, 2023-04-25:
I am having this issue also. I am using Fluid on the Sonic Pad. It shows the thumbnail on the Sonic Pad but not in Fluid.

Sofronio, 2023-04-26:
Same here. No picture, and layer number stuck on 0 of 0.

CarpeNecopinum, 2023-04-26:
I have tracked this down using some older files where the preview worked.
The older files contained the following lines between the `HEADER_BLOCK` and the `THUMBNAIL_BLOCK`: 

```
; hack-fix: write fake slicer info here so that preprocess_cancellation can process.
; generated by SuperSlicer 1.5.0 on 2023-04-21 at 13:13:47
```

If I just copy&paste them into the new files, the thumbnail shows up again. This sounds like it's not really OrcaSlicer's "fault" for breaking the thumbnails, yet still the effect only surfaces because emitting those two lines was disabled in a recent version...

hliebscher, 2023-04-26:
@jonathanalevi 

Have you read this?

NOTE:
1. Please check [here ](https://github.com/SoftFever/OrcaSlicer#how-to-install)for installation instructions.
2. If your profiles didn't show up after updating, please restart OrcaSlicer
3. Klipper users, please update Moonraker to the latest version so that it can recognize OrcaSlicer and parse metadata like thumbnails.
4. If you're running Klipper, it's strongly recommended to add the following configuration to your printer.cfg file.

# Enable object exclusion
[exclude_object]

# Enable arcs support
[gcode_arcs]
resolution: 0.1

CarpeNecopinum, 2023-04-27:
@hliebscher I can confirm this works for me. I updated moonraker (to current master) yesterday, but it wouldn't show the thumbnails until I changed the .gcode file via the editor (today; just a trivial edit to cause the file to be considered "changed").

TonyK54, 2023-04-27:
The bug and the "hack" can be confirmed. I pasted the 2 lines in from @CarpeNecopinum. On saving the file the thumbnail showed up and the items layer height, filament usage, filament weight, etc, were displayed. Previously the thumbnail and the data was missing. 

@SoftFever , the 2 lines may need to be emitted again in the gcode that is rendered. 

```
; hack-fix: write fake slicer info here so that preprocess_cancellation can process.
; generated by SuperSlicer 1.5.0 on 2023-04-21 at 13:13:47
```

CarpeNecopinum, 2023-04-27:
>  the 2 lines may need to be emitted again in the gcode that is rendered.

@TonyK54 you don't need the hack anymore if you update Moonraker to the latest version, restart it and reupload your gcode files (or do a trivial change).

TonyK54, 2023-04-27:
Not everyone has or is willing to flash their systems and may have stock FLSun firmware. Hence there is a bug for a lot of people to be resolved. 

> > the 2 lines may need to be emitted again in the gcode that is rendered.
> 
> @TonyK54 you don't need the hack anymore if you update Moonraker to the latest version, restart it and reupload your gcode files (or do a trivial change).

frankt2012, 2023-04-27:
> Not everyone has or is willing to flash their systems and may have stock FLSun firmware. Hence there is a bug for a lot of people to be resolved.

This is true. I use the Sonic Pad and you are unable to update moonraker currently. I have to wait until Creality updates it. All the other slicers work fine with thumbnails on the older moonraker version.

frankt2012, 2023-04-27:
> The bug and the "hack" can be confirmed. I pasted the 2 lines in from @CarpeNecopinum. On saving the file the thumbnail showed up and the items layer height, filament usage, filament weight, etc, were displayed. Previously the thumbnail and the data was missing.
> 
> @SoftFever , the 2 lines may need to be emitted again in the gcode that is rendered.
> 
> ```
> ; hack-fix: write fake slicer info here so that preprocess_cancellation can process.
> ; generated by SuperSlicer 1.5.0 on 2023-04-21 at 13:13:47
> ```
> 
> I have updated file GCode.cpp in method "export_thumbnails_to_file" to put these 4 lines (there is a blank line above and below this block) and will test it later today or early Friday morning.

I can confirm this fixes the thumbnail issue on the Sonic Pad.

hliebscher, 2023-04-27:
@TonyK54 @CarpeNecopinum 
you can add this 2 lines with post processing script....
or write this in start g-code

TonyK54, 2023-04-27:
> @TonyK54 @CarpeNecopinum
> you can add this 2 lines with post processing script....
> or write this in start g-code

We have to update the application C++ code to fix this.

TonyK54, 2023-04-28:
Have submitted a pull request to the "main" branch at https://github.com/SoftFever/OrcaSlicer/pull/878. Hopefully it will be reviewed and added to 1.6.2.

SoftFever, 2023-04-28:
Hi everyone,

I believe that the better solution would be to push printer and pad manufacturers to provide updates. Klipper, Moonraker, Fluidd, and Mainsail are constantly improving, and being locked into an outdated version is the last thing we want in my humble opinion.

As a workaround for those who cannot or do not want to upgrade Moonraker, you can try adding a fake slicer ID in the start G-code.
[image]

TonyK54, 2023-04-29:
Yes, adding them to the starting gcode does the same thing. thank you

blazecoding2009, 2023-05-23:
So I added the lines to my start gcode and it still isn't working. 
[image]
[image]
I have a sonic pad

invidtiv, 2023-09-20:
[image]
just remove empty line and it works

closed 2023-12-27 by github-actions[bot]