OrcaSlicer/OrcaSlicer #843: Aux fan disable >8sec layertime

foxmckraut, 2023-04-24:
the aux fan is necessary for layertimes <8sec to handle but can do warp on big prints with much much more layertime.

Feature reuquest aux fan only runs <8sec layertime(changable?) for pla

f.e. while layertime <8sec do auxfan 70%(depending on cooling in filament setting).
while layertime >8sec do auxfan 0%

goal: no warps and enough cooling when needed.

for abs/asa its not necessary but can run same for 3sec layertime and onla f.e. 30% aux fan

SoftFever, 2023-04-29:
It's a bit tricky cause those large aux fasn have a very long kick-in time.
Though, why not just always turn on aux fan if you are print PLA?

foxmckraut, 2023-04-29:
> It's a bit tricky cause those large aux fasn have a very long kick-in time. Though, why not just always turn on aux fan if you are print PLA?

Causen on big Prints it warps. f.e. big base like 20sec layertime and gots faster on top to like 4sec layertime so the aux could slowly kick in the lower the layer gets or hardly wenn it hits <8sec with a threshold as option f.e. start 10sec before and run 10secs after(if again >8sec layertime)

Photogad, 2023-05-21:
> It's a bit tricky cause those large aux fasn have a very long kick-in time.
> Though, why not just always turn on aux fan if you are print PLA?

@SoftFever In my use case, I think 70% aux fan is good to run continuously for most PLA prints (as this is the 70% Bambu has in their own profiles), but it would be very helpful to kick this fan up to 100% speed while bridging or doing overhangs (and return to 70% for regular printing). This would allow for better bridging and overhangs.

I did also address this feature request with Bambu, but they did not respond yet.

Regarding fan kick-in time, I think the time for fan to turn from off to on is long, but to go from 70% to 100% for example is much shorter and it does it very quickly.

It's possible we could time how long it takes and add a pre-delay time to the gcode.

julie777, 2023-05-23:
For me aux fan is never necessary for PLA and I modify the filament to turn it off.

foxmckraut, 2023-05-23:
> For me aux fan is never necessary for PLA and I modify the filament to turn it off.

aha you can modify your pla to print with 4s layertime? good for you.

closed 2023-11-20 by github-actions[bot]

FlowForm, 2024-07-23:
I just installed a second AUX fan on the right side, and it solved my issues with overhangs and unequal cooling from the standard solution. However I don't wan t 2x full AUX fan on at all times. 

So I would love to see some more control over the auxiliary fan(s). Perhaps so that the "Force cooling for overhangs" controlled the AUX fan instead of the part fan!?

thetic, 2025-02-26:
I am interested in this feature. I'd like more fine tuned controls of the aux fan.