OrcaSlicer/OrcaSlicer #849: Unable to remove user presets (Filament and/or Printer)

rcxrbt, 2023-04-25:
**Describe the bug**
Trying to remove user presets (Filament and/or Printer), this are still presents after software restart

**To Reproduce**

1.  Remove a user presets (Filament and/or Printer)
2.  Ok, presets are removed :)
3.  Restart sofware
4.  The presets are still here :)

This beavior also happen if I try to modify and save the preset whit same name...
In this case, the preset still show old values...

**Expected behavior**
Really remove presets

**Screenshots**
[image]
[image]

**Printer model**
TwoTrees Sapphire

**Desktop (please complete the following information):**
 - OS:  Windows
 - Version 1.6.1

rcxrbt, 2023-04-29:
Trying to export, the slicer export only some presets  of all user presets showed in the menu (as like a gost preset).

Example:
 two preset in menu:
TwoTrees Generic PLA+ 
TwoTrees Generic PLA++

Export only TwoTrees Generic PLA++

[image]

closed 2023-11-20 by github-actions[bot]

diaudio, 2024-01-15:
Me too - cannot figure out how to remove printer

Jax1er, 2024-04-20:
did this ever get figured out

iamlite, 2024-04-27:
also stuck with this.

allanwrench28, 2024-04-27:
Yep this has annoyed me, still cant figure it out

fredluetkemeier, 2024-05-04:
Why is this closed? This is still an issue

drumstick77, 2024-05-04:
try this...if you go to the printer section on top left and click on the edit
[image]

then click on the X
[image]

I think that will deleted your printer's preset along with all the settings for it...

Gurba667, 2024-05-04:
I think you are right. 👍👍

drumstick77, 2024-05-04:
glad can help. I think all of users' presets have got the X that you can click to delete. Not very obvious.....i know, cuz took me a while to find it....lol

fredluetkemeier, 2024-05-04:
> try this...if you go to the printer section on top left and click on the edit 
> then click on the X 

Thank you! This is exactly what I was looking for. Wish they would make this more obvious

Jax1er, 2024-05-14:
> > try this...if you go to the printer section on top left and click on the edit
> > then click on the X
> 
> Thank you! This is exactly what I was looking for. Wish they would make this more obvious

This is fantastic!  Thanks

PolGuixe, 2024-05-20:
I've created a printer config that make Orca Slicer crash. 

Do delete the printer I need to access it and it crashes every time.  It would be nice to be able to delete it without having to access it. 

Thanks

drumstick77, 2024-05-20:
if you're on windows, I believe all profiles are save in your appdata folder --> C:\Users\[username]\AppData\Roaming\OrcaSlicer
I've never delved into this, so I'm not too sure exactly where they are, but perhaps you might be able to find the corrupted profile here and delete it?

xdevs23, 2024-05-26:
> try this...if you go to the printer section on top left and click on the edit [...]

I don't have an X... Just Save and Search. I removed all printers in the "Select/Remove printers(system presets)" dialog and confirmed it but the printers are still there.

fabiovix, 2024-12-02:
Hi!   I can confirm this issue on my Ubuntu 24, version:   OrcaSlicer 2.2.0   (executable:   "OrcaSlicer Linux Ubuntu 2404 V2.2.0.AppImage").  There is no button to delete the printer profile, so once configured, it remains. It's a VERY annoying bug

Boerny41, 2024-12-25:
> I don't have an X... Just Save and Search. I removed all printers in the "Select/Remove printers(system presets)" dialog and confirmed it but the printers are still there.

Same problem here