OrcaSlicer/OrcaSlicer #850: "Send" does not respect changed plate name in dialogue

pandel, 2023-04-25:
Hi!

If you hit the "Send" button after you sliced your file you normally can change the name of the plate in the middle of the dialogue. But this actually does not work. It always sends the gcode named after the original filename even though you gave it a new name in this dialogue. This is not very helpful, let's say, you want to send the plate multiple times with different print settings, only the last one sent will survive.

Perhaps this can be fixed :-) - keep up the good work and many thanks!

Regards,
Holger

SoftFever, 2023-04-28:
@pandel 
I don't quite understand the issue.
What printer are you referring to? 
Some screenshots to explain the problem will be very helpful.

pandel, 2023-04-28:
Ok, here comes the screenshot... as you can see, the 3mf file itself is named "Bambu Spool Holder", but I renamed the plate to "Bambu Spool Holder blabla". When I send the plate, the file is still being named "Bambu Spool Holder" on the SD card, missing my added "... blabla". So atm I am unable to send the same file multiple times with different names.

[image]

SoftFever, 2023-04-29:
@pandel 
Unfortunately, this is BL firmware/cloud service issue.

pandel, 2023-04-29:
Mmmhh... ok. Thanks.

closed 2023-04-29 by pandel