OrcaSlicer/OrcaSlicer #861: Allow custom primitive objects to be imported to the slicer

MiscPrinter, 2023-04-26:
**Is your feature request related to a problem? Please describe.**
I would like the ability to import custom objects to the primitive menu. I watched this [video](https://www.youtube.com/watch?v=YL5lYEXbLDo) about support helpers and want to have them (or similar ones) added to the primitive menu. 

**Describe the solution you'd like**
Allow the user to import model and have them permanently accessible by the slicer in the form of primitives. 

**Describe alternatives you've considered**
Have the slicer ship with some support aid pre-packaged into the primitive menu. 

**Additional context**
I have designed my own versions of the models in the video should they be needed for implementation I can share them. Having a set of support aids built in to the slicer which are accessible to all would be awesome. 

Also having mouse ears built in would be cool. Mouse ears are easy to make using the cylinder primitive so less of an issue.

grantpeterwilliamson, 2023-10-19:
a little late to the party, but i'd love to see this, my usecase is easily adding and calling apon my common klipper calibration files like speed tower and linear advance cube.

i've already made a profile with all settings for both, i just currently have to open a folder and drag it in.

the ability to just grab it from the primitive menu, while not groundbreaking, would be, shall we say, ideal? :)

IntenseWiggling, 2023-11-09:
+1 as a feature that would be super convenient. Have a number of 'primitives' I use regularly for calibrating filaments and maintenance checks. Would be super useful to pull these in with a right click -> add primitive, vs. pulling up the folder I keep them in and dragging / dropping into the slicer.

Currently the available primitives are stored in `resources/handy_models`, but the items in add primitive menu are basically hard-coded.

Ref: https://github.com/SoftFever/OrcaSlicer/blob/main/src/slic3r/GUI/GUI_Factories.cpp#L488-L520

Having the slicer build this menu from the contents of the directory (and/or the contents of the user data directory) would be a useful way to handle this.

ShaneDelmore, 2023-12-30:
I'll take a crack at this.  I haven't contributed to this project before so I don't know if the feature or implementation would be accepted but for a first pass I will scan the user resources/shapes folder for a list of files (as that is where the prusa shapes gallery keeps them) and add them to the handy models menu. To allow users to override the built in shapes if I add a shape of a given name from the user shapes folder I will not add a model from system resources of the same name.

MiscPrinter, 2023-12-30:
> +1 as a feature that would be super convenient. Have a number of 'primitives' I use regularly for calibrating filaments and maintenance checks. Would be super useful to pull these in with a right [...]

I did try adding the desired custom primitive stl to the resource folder where the other primitive objects are stored. Orca did not acknowledge/index those files and allow one to call on them into the prepare environment. I didn't think it would be that simple but one can try.

ShaneDelmore, 2024-01-01:
> I did try adding the desired custom primitive stl to the resource folder where the other primitive objects are stored. Orca did not acknowledge/index those files and allow one to call on them into [...]
 
This works on my branch but overall the feature as requested of iterating through the user data directory to find models to populate the menu was rejected.

ShaneDelmore, 2024-01-01:
I think the desired implementation here is to port Prusa Slicer's shape gallery but that isn't a feature I am up for tackling so I'll leave it to someone else.

closed 2024-04-09 by github-actions[bot]