OrcaSlicer/OrcaSlicer #864: Disable Wipe Feature

Surreality66, 2023-04-27:
The Wipe Feature is causing some serious issues for my products.  
I appreciate the disable wipe on loops feature, it works great but it's not completely solving my problem.  

I make loops, Rings.  The wipe that is performed at the seam turns inward and leaves a sharp point on the inside of my prints.  
For people that print anything tube related it seems that the wipe is going to leave an artifact on the interior of the print at the seam in the first layer.  It's happening on every single one of my parts at least.    
In my case the interior is part of the finished product.  I make tens of thousands of these items and I'm stressing that I can't get this to leave a smooth internal finish on my prints.  Any chance we can disable it completely or make it turn toward the opposite direction?

Pmant, 2023-04-27:
There are settings to disable wiping in the printer -> extruder section aswell as overrides in the filament settings. Don't they disable all wiping?

Surreality66, 2023-04-27:
> There are settings to disable wiping in the printer -> extruder section aswell as overrides in the filament settings. Don't they disable all wiping?

I could freaking hug you.  
Send me your address so I can drive wherever you are and just give you the most epic high-five you've ever had.  

I spent ages looking for just this and I'm not ashamed to say that I'm a doofus for missing it.  

You just saved my ass.  
(lemme just go and do that now)

closed 2023-04-27 by Surreality66