OrcaSlicer/OrcaSlicer #867: Use "Motion ability" only for time estimate

rfnovo, 2023-04-27:
Add the option to use Motion ability numbers only to estimate print time, instead of always emitting to Gcode, similar as we have in PS and SS: 
[image]

mikebcbc, 2023-04-29:
+1 to this!

bonjorno7, 2023-08-10:
I would like to expand this idea to include not just machine limits, but speed/acceleration/jerk settings in print profiles too.
Preferably as separate options, because I would emit speed settings to G-code, and use acceleration and jerk only for time estimate.

closed 2023-11-20 by github-actions[bot]

bonjorno7, 2023-11-20:
I didn't think the bot would actually close it.
I this would still be a nice feature and should remain open until someone implements it.