OrcaSlicer/OrcaSlicer #869: Hide GCODE preview when no in use

rfnovo, 2023-04-27:
When previewing the layers, the GCODE windows is hidden by default:
[image]

And it will appear whenever the horizontal bar is touched, to show the specific GCODE:
[image]

But once this GCODE preview window appears it will no longer disappear, as it happens in other slicers where moving the vertical layer bar will make the GCODE preview window auto-hide:
[image]

SoftFever, 2023-04-28:
Hi @rfnovo 
you can press `c` hotkey to hide it

closed 2023-04-28 by rfnovo

Dynom, 2023-10-27:
Hi,

I'd like to re-open this issue. As the OP states, albeit a bit cryptic, the Gcode window only appears when clicking on the horizontal slider. The hotkey-C shortcut doesn't show/hide the window, prior to clicking this slider.

Dynom, 2023-11-01:
@rfnovo ping

Shaolo, 2023-12-08:
This hotkey should be added to the view menu.