OrcaSlicer/OrcaSlicer #870: Bug missing file Snapmaker J1

Frelidon, 2023-04-27:
**Describe the bug**
The following text is displayed: The configgured post-processing script does not exist: /Users/hliebscher/smfix_bs-darwin-arm64

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.
https://drive.google.com/drive/folders/1nhNQly_PCtsUed0oq-DFHpmh4_tONGHy?usp=sharing

**To Reproduce**
eliminate error !

**Expected behavior**
A clear and concise description of what you expected to happen.
Would be nice if you can fix it would like to use the OrcaSlicer because it is very easy to use.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Printer model**
Snapmaker J1

**Desktop (please complete the following information):**
 - OS: [Windows11 & Linux Mint]
 - Version [W11 & Mint21.1LTS]

hliebscher, 2023-04-27:
its not a bug you have to select your own script....
you can read here 

https://github.com/SoftFever/OrcaSlicer/discussions/782

https://github.com/SoftFever/OrcaSlicer/pull/663

you find all files that you need here pls select your os version...

https://github.com/hliebscher/Snapmaker2Slic3rPostProcessor/releases

Frelidon, 2023-04-28:
Hi hab gesehen du bist auch aus Deutschland. Kannst du mir bitte per Discord erklären was ich machen muss. Geht immer noch nicht.

closed 2023-04-28 by Frelidon

MarvDarf, 2023-07-30:
Hallo, hab bereits einige Stunden damit verbracht Orca Slicer zum laufen zu bekommen . Aber ich verzweifel leider daran .Hätte jemand der Deutsch kann vllt eine gute Anleitung oder Zeit zum Telefonieren bzw. Um mir bei meinen Problemen zu helfen ?