OrcaSlicer/OrcaSlicer #879: Pressure Advance calibration doesn't work with 0.2nozzle for PLA

zviratko, 2023-04-28:
**Describe the bug**

I thought the problem was changing a slot in AMS makes the speed constant - turns out it's because of volumetric flow limit which makes it constant at 2mm^3/s.

When doing the PA calibration (Lines method), it is not possible to use PLA with a 0.2mm nozzle due to volumetric speed limit.
Once PLA is used the speed becomes constant.
I think the issue lies in the volumetric limit of 2mm^3/s with PLA, changing it higher or lower makes the speed vary.

Interestingly it seems to "work" with PETG, which has a volumetric limit of 1mm^3/s, but I'm wondering if it actually results in useful calibration, I suspect not.

I'm wondering if the volumetric limit makes it pointless to calibrate. The built-in X1C flow calibration does make a difference (though I don't think for the better as the resulting "calibrated" line is bloby and ugly), but I don't think it is made for 0.2mm nozzle.

So I guess the questions are
1. Is it useful to calibrate for 0.2mm/s nozzle? 
2. Is it useful to calibrate with the default volumetric limit? (Should the calibration override it?)
3. Is it useful to calibrate with a higher limit then changing it back?
4. Is the default limit real? I think it should be able to perform better than 2mm^3/s for PLA, but I haven't tested it and this is what comes as default
5. Should there be logic in there which detects that the calibration is not going to be useful, either because of volumetric flow limit or too slow speed for any other reason?

Thanks

zviratko, 2023-04-28:
Volumetric flow set to 10 mm^3/s
[image]

Default volumetric flow of 2 mm^3/s
[image]

Default of PETG - 1 mm^3/s
[image]

Tibaka, 2023-10-14:
Hi, i jump into your post because I was facing the same problem. Regardless of the nozzle size, I'm sure the PA need to be calibrated. 

I don't know why but, in Bambu slicer or Orca slicer, if I use the 'X1C 0.2 nozzle' default printer profile, I can tweak all I want, it will result the same thing, the Max Volumetric speed can't go above 1mm3/s ( I tested only with PLA). 

However, if I use the 'X1C 0.4 nozzle' default printer profile and tweak everything to match my 0.2 nozzle. It works like a charm.

Something is forced in the printer profile and can't be access from the user interface ?

Best,
Tibaka

closed 2024-01-21 by github-actions[bot]

netweaver1970, 2024-04-23:
all these in-build calibrations are only really working for 0.4mm nozzles due to (probably) plenty of hardcoding while generating. I had the same issue for other calibrations, for bigger nozzles. That was also never looked into by the devs, probably because of bigger fish to fry. 
Probably rightfully so, seeing how unstable things sometimes can be :)