OrcaSlicer/OrcaSlicer #880: Plate name says "Untitled" in orange

chuckkay, 2023-04-28:
**Describe the bug**
Each plate has a plate number in in the front right corner, but the the back left corner has the word "untitled"
the 3md file is saved with a name, the objects and assembly's have names, I even went to the project tab and gave the model a name, yet i cannot seam to find anyway to get rid of "Untitled" 

**x1**

**Desktop (please complete the following information):**
 - OS: windows 11
 - Version orca 1.6.2 beta
[image]

SoftFever, 2023-04-28:
You can find more information about plate name here: https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.6.0

closed 2023-04-28 by SoftFever

chuckkay, 2023-04-28:
The simpler answer was “hit the gear icon at the corner of the plate”