OrcaSlicer/OrcaSlicer #885: Add option to start cooling fan speed increases earlier than when needed (compensate for spool-up time), and disable cooling fan speed changes for short overhangs

scottmudge, 2023-04-28:
**Is your feature request related to a problem? Please describe.**
When printing quickly (>150mm/s), when approaching an overhang or bridge where a high cooling fan speed is required, often the fan will take a second or two to spool up to it's required speed. For some overhangs, it might already be half-way through printing the overhang before the fan is up to speed.

My idea is to add an option to start the fan speed increase a certain number of printed millimeters earlier than it is required (or better yet, seconds of print time, but that might be hard to implement), to compensate for the spool-up time.

It might also be useful to add an option to disable fan-speed changes for short overhangs, which take less time to print than the time it takes for the fan to even speed up.

tome9111991, 2023-04-29:
Looks like:
https://github.com/SoftFever/OrcaSlicer/issues/143

scottmudge, 2023-04-29:
> Looks like: #143

Ah that is basically exactly it! Would be nice to add a "Disable cooling fan changes for short overhangs" option while at it.

I might make a PR for this, but I'm having trouble compiling OrcaSlicer w/ VS 2022. Which is strange because PS and SS both compiled fine.

scottmudge, 2023-04-30:
Just making a note here that I got OrcaSlicer compiling now with VS2022. Had to update Boost to 1.78, and moved the wxWidgets  patch which was being applied by CMake to a static git tag on a fork here (https://github.com/scottmudge/wxWidgets-bambustudio/tree/v3.1.5-bambu), since my version of CMake (latest) was throwing a bunch of errors if attempting to rebuild the dependencies, since it was trying to re-apply the patch without resetting the files.

Makes the dependency build a bit easier, and it's good to have Boost updated a bit.

Changes for that here: https://github.com/scottmudge/OrcaSlicer/commit/e07b39e88be085012ffd3cc1b49157ab6f852406

I'm now working on porting over the fan speed delay and fan speed kickstart times from SuperSlicer to OrcaSlicer. The only limitation (as noted in SuperSlicer) is that it will really only work with G0 and G1 gcode commands, and not on Arc gcode commands, since it would require splitting the preceding arc at a distance equivalent to the fan speed delay * arc travel speed.

scottmudge, 2023-04-30:
Alright got it ported over, required some changes from SuperSlicer. Mainly SS has the option for percentage-based fan speeds (rather than the 0-255 for PrusaSlicer/OrcaSlicer/etc), so I had to remove that. And some toolchanger code SS had, which didn't really apply to OS.

Seems to be working, changes here:

https://github.com/scottmudge/OrcaSlicer/commit/1af78eee02e7664b557f15d2a897a299709201e0

Example:

[image]

Coloring is fan speed. Green arrow points to the fan speed being increased before the overhang starts. Increasing the `Fan speed delay` parameter means this fan speed start moves farther back before the start of the overhang.

Will do more testing and create a PR for it.

scottmudge, 2023-05-01:
Alright put in a PR for this, #910

closed 2023-05-01 by scottmudge

florianmuetzel, 2023-05-09:
Thanks for that. Was searching for that feature ;-)