OrcaSlicer/OrcaSlicer #886: KDE - display global scale above 150%

adolfotregosa, 2023-04-28:
With KDE display global scale above 150% orca-slicer will not open. Terminal shows the following error. 

```
gtk_widget_set_size_request: assertion 'width >= -1' failed
Fatal glibc error: malloc.c:2593 (sysmalloc): assertion failed: (old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)
Aborted (core dumped)
```

pujvz, 2023-06-18:
This is unrelated to KDE, however it is related to scaling. I use a window manager and orcaslicer crashes with the exact same error when `Xft.dpi: 192`. orcaslicer does not crash with this error when `Xft.dpi: 96`.

orcaslicer is unusable on a high dpi monitor without scaling.

OS: archlinux

pujvz, 2023-08-07:
> Any solution / workaround for that?

Nope. I ended up running it in windows instead.

closed 2023-11-20 by github-actions[bot]

mabl, 2024-01-04:
Se also https://github.com/bambulab/BambuStudio/issues/1868.