OrcaSlicer/OrcaSlicer #887: Post-processing scripts don't work Snapmaker J1

Frelidon, 2023-04-28:
It doesn't work on the Snapmaker J1, no matter which version I use. Sometimes he doesn't find it. Sometimes the script doesn't start. Please make a video tutorial. Would like to use the slicer.
In Cura and Luban it works without any problems.

Greetings Frelidon.
From Bremen Germany
[image]
[image]

SoftFever, 2023-04-29:
Hi @Frelidon 
This thread may help you: https://github.com/SoftFever/OrcaSlicer/discussions/782

Frelidon, 2023-04-29:
Thanks, I already know that, unfortunately it doesn't help.

Frelidon, 2023-04-29:
Thanks, I already know that, unfortunately it doesn't help.

> Hi @Frelidon This thread may help you: #782

closed 2023-04-29 by Frelidon

hliebscher, 2023-04-29:
@Frelidon 

Sorry for German

Du benutzt die falsche Version. Alles mit Darwin ist für mac.
Du musst die richtige Version für Windows aus meinem GitHub laden ist unter Q&A unter Diskussionen erklärt runterladen. 
Und immer die Version mit BS im Namen sonst geht das nur für PrusaSlicer
Und zum senden kannst du sm2uploader nehmen 
Sonst können wir auch direkt in Kontakt treten 
Email von mir in der deutschen po Datei im GitHub 

Wir bekommen das hin

hliebscher, 2023-04-29:
You are using the wrong version. Everything with Darwin is for mac.

You have to download the right version for Windows from my GitHub is under Q&A under Discussions explained download.

And always the version with BS in the name otherwise it only works for PrusaSlicer

And to send, you can use sm2uploader

Otherwise, we can also get in touch directly

Frelidon, 2023-04-29:
> @Frelidon
>
> Sorry for German
>
> Du benutzt die falsche Version. Alles mit Darwin ist für mac. Du musst die richtige Version für Windows aus meinem GitHub laden ist unter Q&A unter Diskussionen erklärt [...]

Hi kein Probelm bin auch aus Deutschland (Bremen) kannst mir gerne auf Discord oder Steam schreiben. Frelidon Heiße ich über-all.