OrcaSlicer/OrcaSlicer #889: Profile deletes with Post Processing Script

frankt2012, 2023-04-29:
**Describe the bug**
When I add a post processing script my profile is deletes on restart.

**To Reproduce**
Create profile and add post processing script. Profile will be deleted when Orca restarts.

**Expected behavior**
It should save it.

**Printer model**
Ender S1 with Sonic Pad

**Desktop (please complete the following information):**
Windows 11

SoftFever, 2023-04-29:
Can you find the process file(.json) under %appdata%/OrcaSlicer/user/*/process?
If so, can you share it?

frankt2012, 2023-04-29:
> Can you find the process file(.json) under %appdata%/OrcaSlicer/user/*/process? If so, can you share it?

Here it is. Works fine when I make the profile. But as soon as I exit the slicer and restart it instantly deletes itself.

[Ender S1 Klipper 0.20mm Standard @MyKlipper.zip](https://github.com/SoftFever/OrcaSlicer/files/11359320/Ender.S1.Klipper.0.20mm.Standard.%40MyKlipper.zip)

SoftFever, 2023-04-29:
@frankt2012 
It seems that the double quote is causing trouble with the JSON format. As a workaround, could you put the "klipper_extimator.exe" file in a path that does not contain any whitespaces? 
Then in OrcaSlicer, don't double quote the path:
[image]

This should fix you problem

frankt2012, 2023-04-29:
> @frankt2012 It seems that the double quote is causing trouble with the JSON format. As a workaround, could you put the "klipper_extimator.exe" file in a path that does not contain any whitespaces? Then in OrcaSlicer, don't double quote the path: [image]
> 
> This should fix you problem

I will try that. The only reason for the quotes was because I couldn't get it working without. Will try again.

frankt2012, 2023-04-29:
> @frankt2012 It seems that the double quote is causing trouble with the JSON format. As a workaround, could you put the "klipper_extimator.exe" file in a path that does not contain any whitespaces? Then in OrcaSlicer, don't double quote the path: [image]
> 
> This should fix you problem

Ok that worked. I don't know why but I thought I did that in the beta. Thank you.

closed 2023-04-29 by frankt2012