OrcaSlicer/OrcaSlicer #890: Klipper Moonraker

drveni, 2023-04-29:
in the appendix of the text on "4. If you're running Klipper, it's strongly recommended to add the following configuration to your printer.cfg file."

to insert additional values

[gcode_arcs]
resolution: 0.1

this does not work in a way that the print stops by itself. the printer in the klipper reports an error that the print is paused. pauses occur randomly. without any rule at what height or time. a paused print can only be continued/resumed once. after the second pause, the printer simply freezes and does not want to print anymore. there is nothing in the klipper log, everything is empty, just like in the moonraker log. as if nothing had happened.

when I delete those two lines from printer.cfg, the printer works without any problems. the printer uses the motherboard from ender 3 v2 4.2.7. it is a monified printer. 

I don't know if there is anyone else with this bug. but for me it doesn't work as it should according to the mentioned documentation. I would like to mention that this is probably in my case in combination with that motherboard, because if I use another slicer program, the bug repeats itself.

if there is a possibility to somehow implement or specify somewhere and that it is additionally tested by other users who have that motherboard. so if it is true with this problem, to write in the documentation as an additional note

fmonaca, 2023-04-29:
I am experiencing a similar issue. It's not the [gcode_arcs] section on Klipper the issue though. I had it already active in printer.cfg and it works without issues with other slicers. It is more the "Arc  fitting" option in the slicer. If selected, I have Klipper crashing due to a "klipper MCU 'mcu' shutdown: Timer too close". If I deselect the option in the slicer, it prints with horrible quality on curved surfaces. I have tried other slicers (PS 2.6 Alpha and Cura 5.2) and they work without crashing Klipper and with nice curved surfaces finish.

SoftFever, 2023-04-29:
> in the appendix of the text on "4. If you're running Klipper, it's strongly recommended to add the following configuration to your printer.cfg file."
>
> to insert additional values
>
> [gcode_arcs] [...]

@drveni, I have disabled the arc fitting option by default in most built-in profiles in Orca Slicer. It is perfectly fine if you do not add gcode_arcs to the Klipper configuration if you encounter any issues. Since Klipper internally interpolates arc moves into multiple small straight moves (g1), it will not have any impact on printing quality. The benefit of enabling this option is that the resulting gcode file is smaller.

The arc commands themselves work fine for most users, including myself. However, I suggest updating Klipper to the latest version if you haven't already. I'm not sure what MCU you are using, but an older or slower 8-bit MCU may not be powerful enough to real-time interpolate commands at this resolution (0.1mm). This is just my guess.

SoftFever, 2023-04-29:
> I am experiencing a similar issue. It's not the [gcode_arcs] section in Klipper the issue though. I had it already active in printer.cfg and it works without issues with other slicers. It is more the "Arc fitting" option in the slicer. If selected, I have Klipper crashing due to a "klipper MCU 'mcu' shutdown: Timer too close". If I deselect the option in the slicer, it prints with horrible quality on curved surfaces. I have tried other slicers (PS 2.6 Alpha and Cura 5.2) and they work without crashing Klipper and with nice curved surfaces finish. In the photo: above is with Cura 5.2, below is with Orcaslicer 1.6.2 Win64 with Arc fitting option not selected. I was not able to print the part with Orcaslicer selecting the Arc fitting option as it would crash Klipper around 20% of the print.
> 
> [image]

What resolution did you set in Klipper for the arc_gcode Klipper?

drveni, 2023-04-29:
> @drveni, I have disabled the arc fitting option by default in most built-in profiles in Orca Slicer. It is perfectly fine if you do not add gcode_arcs to the Klipper configuration if you encounter [...]

thanks for the replay

how to turn off in orca slicer? where exactly is it located? is that a resolution? i need put to 0?

im print from superlicer and prints result are great and clean. 

mcu STM32F103 28bit loader As I remember. once I flashed it, I didn't touch it again and it works. and I haven't seen any better or worse print quality in orcaslicer. it just freezes when it's turned on in my printer.cfg.

SoftFever, 2023-04-29:
> thanks for the replay
>
> how to turn off in orca slicer? where exactly is it located? is that a resolution? i need put to 0?
>
> im print from superlicer and prints result are great and clean.
>
> mcu [...]

Resolution is configured in your Klipper printer.cfg, it has nothing to do slicer.

> [gcode_arcs]
> **resolution**: 0.1

Anyway, to avoid generating any arc move instructions in OrcaSlicer, make sure
You don't tick the Arc Fitting option in Process settings:
[image]
And don't select Spiral in Printer settings
[image]

fmonaca, 2023-04-29:
> 
It is 0.1. It is weird that not using Arc fitting produces those artifacts on curved surfaces. 
I am using an Octopus Pro with a Rpi 3B.

ytugarev, 2023-07-05:
I also have issues with "klipper MCU 'mcu' shutdown: Timer too close" on klipperized Sovol SV06+. It has 32-bit MCU so I'm not sure it is compute issue. Does it matter what clock reference I chose?
[image]

I ended up disabling it completely but it did not worsen quality 
@drveni did you change "Precision\Resolution" from default 0.012 mm to something else?

stofancy, 2023-10-02:
I'm facing exactly same issue too, and disable `arc_fitting` will fix it.
I'm using BTT M8P as printer controller board + use CB1 as SOC. 
[image]
the performance of CB1 SOC is very similar like pi 4.
Klipper config:
```
[gcode_arcs]
resolution: 0.1
```
OrcaSlicer precision:
[image]
@SoftFever , any idea why it happen?

drveni, 2023-10-02:
still didn't fix the problem the bug still exists, I just stopped using orca, went back to superslicer and disabled this option. this just doesn't work for me. its freeze the printer. there is no point in me wasting time on a hobby or a hobby; discovering software problems because of the programmers they simply shift the responsibility that it is not their problem either from orca klipper or eeprom.(that question was also asked on klipper github) my solution for that bug fix... I stopped using it. as much as the slicer has maybe a couple of interesting things, it is just as unusable because they don't know how to set the exact time for the length of the print, there is no option time estimation compesation.

in the meantime, I'm following this post, maybe a miracle will happen and it will work.

stofancy, 2023-10-04:
@drveni , there is a wrokaround, you can increase the resolution in `printer.cfg`:
```
[gcode_arcs]
resolution: 0.2 ; or larger. default is 0.1
```
In my case, the printer will be stuck in to `timer too close` issue immediately at speed 350mm/s when printing arc lines. I increased the resolution from 0.1 to 0.25, now I can print more faster, the cylinder printing test(with concentric infill/top/bottom) now stable at speed 600mm/s.

drveni, 2023-10-04:
> @drveni , there is a wrokaround, you can increase the resolution in `printer.cfg`:
> 
> ```
> [gcode_arcs]
> resolution: 0.2 ; or larger. default is 0.1
> ```
> 
> In my case, the printer will be stucked in to `timer too close issue` immediately at speed 350mm/s when printing arc lines. I increased the resolution from 0.1 to 0.25, now I can print more faster, the cylinder printing test(with concentric as infill/top/bottom) now stable at speed 600mm/s.

Thx for info.I will test during the next week and see. For me, the printing speed is up to 300mm because the extruder can't do more

SoftFever, 2023-10-04:
> I'm facing exactly same issue too, and disable `arc_fitting` will fix it. I'm using BTT M8P as printer controller board + use CB1 as SOC. [image] the performance of CB1 SOC is very similar like pi [...]

Essentially, the issue arises because Klipper struggles with the added workload. 
When Klipper tries to interpolate G2/G3 commands into G1 commands in real-time, and if the Pi's CPU can't process this quickly enough, errors occur. 
There's not much Orca can do in this scenario. 

We have a few options:
- Disable the Arc fitting.
- Adjust the resolution in the Klipper config.
- Upgrade the hardware.

It's worth noting that Arc fitting doesn't offer significant advantages with Klipper firmware, as it doesn't natively support G2/G3 commands. 
The primary benefit of using arc fitting for Klipper setups is the reduction in gcode file size. Which shouldn't matter much for most of us.

SoftFever, 2023-10-04:
There are other tricks like increase Klipper's timeout threshold by modifying TRSYNC_TIMEOUT in it's source codes.
But to be honest, if it's just arc fitting causing the headache, just turn off it.

closed 2024-01-10 by github-actions[bot]

PEZ18, 2024-04-28:
@SoftFever  I was experiencing this issue with my walls, and discovered the cause is quite strange... Maybe you can have a look at my photos and let me know if this is a bug in the slicer, or my settings are just wrong in klipper? I have gcode_arcs enabled in my printer.cfg, and I tried  different resolutions, but nothing made the artifacts go away... Thats when I did some more digging and found the cause.

First I printed a sliced model from PrusaSlicer 2.6 and the walls were perfect. Then I compared gcode from that file to OrcaSlicer 2.0 and discovered that OrcaSlicer was changing the SQUARE_CORNER_VELOCITY  in the gcode before wall prints, but PrusaSlicer didn't have that anywhere in there. 

Next test I set all the jerk settings to the same number, and the gcode no longer had that SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY in there, and the artifacts were completely gone.

Seems like the issue only happens if your wall jerk settings are different from the rest, but why would the walls look this bad from changing the wall jerk higher or lower from the rest? 

Here are some photos of models I printed with various settings: 

If wall jerk is different from rest, it sets the SQUARE_CORNER_VELOCITY to that value and introduces the artifacts in the walls:

[image]

If wall jerk is the same as the rest, there is no SQUARE_CORNER_VELOCITY  in gcode, and there are no artifacts.
[image]

PrusaSlicer 2.6:
[image]

OrcaSlicer 2.0 With Arc, and With Wall Jerk 8:
[image]

OrcaSlicer 2.0 No Arc, and With Wall Jerk 8:
[image]

OrcaSlicer 2.0 No Arc, Wall Jerk all set to 9:
[image]

OrcaSlicer 2.0 With Arc, Wall Jerk all set to 9:
[image]

PEZ18, 2024-05-01:
Should I just start a new topic instead of posting here?

loutwice, 2024-05-07:
same here https://github.com/SoftFever/OrcaSlicer/issues/5297

PEZ18, 2024-05-08:
> same here #5297

I got that issue when I was troubleshooting and tried increasing the resolution of the arc fitting settings in the printer.cfg, but I could not confirm if they are the same or related issues... I feel like this is more of a bug in the slicer caused by the corner velocity change, but I need someone smarter to confirm or deny it.

If you want a simple way to avoid that MCU warning, just decrease the arcs_gcode resolution till it goes away. I currently have mine at .08, but yours might be different? What is yours currently at btw?

[image]

loutwice, 2024-05-08:
> I got that issue when I was troubleshooting and tried increasing the resolution of the arc fitting settings in the printer.cfg, but I could not confirm if they are the same or related issues... I [...]

i have mine at 0.2 in printer.cfg 
may be the arc fitting default is too much in orca slicer?