OrcaSlicer/OrcaSlicer #891: Enable "Change Filament" Option when printing in Spiral Vase mode

LidorBaum, 2023-04-29:
When in preview mode, after slicing a model in vase mode, in PrusaSlicer it's possible to press on the + icon in the right layers-slider and add a filament change command.
In Bambu Studio and Orca Slicer this feature is disabled. Is there an option to make it enabled?
Its working pretty good on prusa Mk3s+.
[image]
[image]

coffeemade, 2023-05-10:
Adding to this as I noticed this same thing, even when not in Vase mode.

I had to disconnect my ams from x1 C and via side loaded filament, I was not able to do a simple layer height filament change. I had a small part I printed multiple times using these options all turned to a single color item:
-Filament 1 set to first section then layer 15 change filament to Filament 2 (no pause, just single color printed)
-Filament 1 set to first section then layer 15 change filament to Filament 1 (no pause, just single color printed)
-Filament 1 set to pause, but then I was unable to change filament on the printer or via handy app. (I didn't test in the slicer)

I also tried in bambu labs slicer and was NOT able to make this happen as well. I have the default filament start and end gcode.

Th3phan, 2023-06-10:
Following this, would love to do multicolour vases!

Calzune, 2023-06-24:
Im also waiting for this... 
[image]

Fire-, 2023-11-13:
I'd also like to see this - primarily so that I can add any sort of gcode as a marker to merge vase and non-vase print gcode. Also not entirely sure I understand why it's intentionally disabled to begin with?

trankillity, 2023-11-15:
Also keen for this. Just doing a custom M600 currently, however that doesn't handle different material types/temps/flow rates.

GuzziRaz, 2023-12-01:
> I'd also like to see this - primarily so that I can add any sort of gcode as a marker to merge vase and non-vase print gcode. Also not entirely sure I understand why it's intentionally disabled to begin with?

Presumably because spiral mode is a single continuous line from bottom to top. Any pause would result in artefacts. Maybe it should just be a warning though, so one could experiment with it.

Eldenroot, 2024-01-21:
> Presumably because spiral mode is a single continuous line from bottom to top. Any pause would result in artefacts. Maybe it should just be a warning though, so one could experiment with it.

Yes, I think this is the way...

JonBerlin, 2024-06-10:
Is there any update or maybe a workaround? It is possible to slice in spiral vase mode and then change the gcode file to include the color change, but it is a slow and somehow painful workflow.

LouisH98, 2024-06-23:
This really should be supported. 
PrusaSlicer supports inserting colour changes in vase mode. It would be great to combine this with the simplicity of the AMS. 
Currently I can do this with a MK3s smoother than a Bambu lab with AMS...

Fire-, 2024-09-22:
commenting to avoid staleness

m9tdev, 2025-08-03:
Found a workaround for some usecases: https://github.com/bambulab/BambuStudio/issues/1002