OrcaSlicer/OrcaSlicer #894: Flush options disabled when Prime Tower is disabled

Desterly, 2023-04-29:
**Describe the bug**
When performing a multi-color print, disabling the prime tower also disables the "Flush Options" .   Enabling Prime tower enables these.

This seems to be new to 1.6.2

**To Reproduce**
Enable/Disable Prime Tower Option and see Flush options enabled/disabled

**Expected behavior**
Prime Tower and Flush Options should be separately controllable

**Screenshots**
[image]
[image]

**Printer model**
Bambulabs Carbon X1 

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.6.2 of the slicer

DHernan, 2023-08-10:
Did you ever find out if this is on purpose?

Desterly, 2023-08-14:
> Did you ever find out if this is on purpose?

It's the same in BambuStudio as well so not sure if it's on purpose but it's an upstream item.  I didn't dig too far though to see if it changes the GCODE when you deselect the prime tower though.

closed 2023-11-21 by github-actions[bot]

Sod-Almighty, 2025-06-04:
Why is this bug closed? It is still happening.