OrcaSlicer/OrcaSlicer #895: [Feature Request] Post-Processing scripts as part of preview pane

mikebcbc, 2023-04-30:
**Is your feature request related to a problem? Please describe.**
Post-process scripts are not running when previewing, only when exporting or sending to printer. This makes things like klipper_estimator or other scripts that change metadata useless as it's not show as part of the slicer preview panel.

**Describe the solution you'd like**
The ability to choose to run the post-processing scripts as part of "previewing" the sliced model would be great.

**Describe alternatives you've considered**
Alternatives are just sending it to the printer and seeing the output there :(

hliebscher, 2023-05-02:
i use this to send the g-code to the printer ... so in preview all file go to printer...
in prusa or other slicers are the same

bonjorno7, 2023-08-10:
I agree it would be very nice to see klipper estimator output inside the slicer before exporting.

closed 2023-11-21 by github-actions[bot]