OrcaSlicer/OrcaSlicer #897: Labels bigger than buttons (German language)

K2ij3, 2023-04-30:
**Describe the bug**
When the german language is selected, the labels are bigger than the buttons (dialog sending print)

**3mf File for This Bug**
not relevant

**To Reproduce**
Select dialog for sending print

**Expected behavior**
Labels fit in the buttons

**Screenshots**
[image]

**Printer model**
X1-Carbon

**Desktop (please complete the following information):**
 - Windows 10
- 1920x1080 resolution
 - Version 1.6.2

hliebscher, 2023-05-01:
can you send a screen capture form english language ... i have no bambu X1

K2ij3, 2023-05-01:
Hello together, english language: Mit freundlichen GrüßenDominik Schiffner Von: Heiko LiebscherGesendet: Montag, 1. Mai 2023 11:30An: SoftFever/OrcaSlicerCc: K2ij3; AuthorBetreff: Re: [SoftFever/OrcaSlicer] Labels bigger than buttons (German language) (Issue #897) can you send a screen capture form english language ... i have no bambu X1

hliebscher, 2023-05-01:
@SoftFever can we fix with dynamic button size because the Labels are on different points in the GUI i can not make  shorter...

K2ij3, 2023-05-10:
Hello, 
I've take a look into OrcaSlicer_de.po
with this translation the button must be very big:

[image]

K2ij3, 2023-05-11:
Solved in 1.6.3 beta