OrcaSlicer/OrcaSlicer #900: Can't compile on macOS

Guilouz, 2023-04-30:
**Describe the bug**
CMake Error at src/libslic3r/CMakeLists.txt:445 (find_package):
  By not providing "FindOpenCASCADE.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "OpenCASCADE", but CMake did not find one.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.4
 - Xcode, Cmake, git, gettext are installed

hliebscher, 2023-05-01:
witch processor type ?

lrsweet, 2023-05-07:
The same error occurs when compiling on Windows 11 using Visual Studio 2019. 

CMake Error at src/libslic3r/CMakeLists.txt:447 (find_package):
  By not providing "FindOpenCASCADE.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "OpenCASCADE", but CMake did not find one.

  Could not find a package configuration file provided by "OpenCASCADE" with
  any of the following names:

    OpenCASCADEConfig.cmake
    opencascade-config.cmake

  Add the installation prefix of "OpenCASCADE" to CMAKE_PREFIX_PATH or set
  "OpenCASCADE_DIR" to a directory containing one of the above files.  If
  "OpenCASCADE" provides a separate development package or SDK, be sure it
  has been installed.

SoftFever, 2023-05-08:
Have you checked the build log deps?
`build_release_macos.sh -d` to build deps

Guilouz, 2023-05-09:
@SoftFever This is the full logs for building deps : 

[logs.txt](https://github.com/SoftFever/OrcaSlicer/files/11430690/logs.txt)

patofoto, 2023-06-29:
Any help solving this would be greatly appreciated

[sh build_release_macos.txt](https://github.com/SoftFever/OrcaSlicer/files/11908279/sh.build_release_macos.txt)

patofoto, 2023-06-29:
@SoftFever Thank you but still not working.
[sh build_release_macos_2.txt](https://github.com/SoftFever/OrcaSlicer/files/11909394/sh.build_release_macos_2.txt)

patofoto, 2023-06-29:
This is what I changed
[OCCT.txt](https://github.com/SoftFever/OrcaSlicer/files/11909410/OCCT.txt)

closed 2023-06-30 by SoftFever