OrcaSlicer/OrcaSlicer #902: Feature request Arc Overhang

TechDude23, 2023-04-30:
Can this be added to the Orca Slicer [ Arc Overhang ](https://github.com/rvmn/SuperPleccer)

scottmudge, 2023-05-09:
Working on porting this on my fork: https://github.com/scottmudge/OrcaSlicer/tree/feature_arc-overhangs

No promises yet -- there are significant differences between OrcaSlicer and SuperPleccer, which may require some debugging/modifications. 

If I get something working, I'll submit a PR

GuzziRaz, 2023-08-30:
Dupe of #409

kostecky, 2023-12-23:
Any luck on this?

starschild86, 2024-01-22:
Would also be very interested in knowing when this will be available

Rickthebig, 2024-03-20:
Checking this topic almost every day...
Arc overhang **urgently** needed!

pagdot, 2024-03-20:
@Rickthebig 
You can use https://github.com/Kelsch/arc-overhang-orcaslicer-integration in the meantime. It is a PoC implementation based on the on the prusa slicer integration which is based on the original script. There are both forks of SuperSlicer (forked of PrusaSlicer) and OrcaSlicer, but development of both seems to have stopped months ago. I have already found a major issue with the current implementation(s) with it starting the arcs on a not supported edge. IMO it is a nice PoC and I'd like to see it integrated in slicers in the future, but the algorithm isn't there yet

Rickthebig, 2024-03-20:
> @Rickthebig You can use https://github.com/Kelsch/arc-overhang-orcaslicer-integration in the meantime. It is a PoC implementation based on the on the prusa slicer integration which is based on the [...]

Thanks, hope dies the last.
I tried to use it as a postprocessor but as I am using Windows 10 right now, and some libraries are needed that are not available for Windows out-of-the-box, python 3.9 throws an error 1.
Quote from https://shapely.readthedocs.io/en/latest/installation.html:
"We do not have a recipe for Windows platforms."
So currently the postprocessor seems to be limited to Linux users.
Correct me If I am wrong.

Rickthebig, 2024-03-20:
Wait - I did find a shapely library for Windows:
https://pypi.org/project/shapely/#files
I will let you know the outcome.

Rickthebig, 2024-03-20:
Unfortunately the script cannot be executed:
from shapely import Point, Polygon, LineString, GeometryCollection, MultiLineString, MultiPolygon
ModuleNotFoundError: No module named 'shapely'

The library was successfully installed:

 pip install shapely
Defaulting to user installation because normal site-packages is not writeable
Collecting shapely
  Downloading shapely-2.0.3-cp312-cp312-win_amd64.whl.metadata (7.2 kB)
Collecting numpy<2,>=1.14 (from shapely)
  Downloading numpy-1.26.4-cp312-cp312-win_amd64.whl.metadata (61 kB)
     ---------------------------------------- 61.0/61.0 kB 461.8 kB/s eta 0:00:00
Downloading shapely-2.0.3-cp312-cp312-win_amd64.whl (1.4 MB)
   ---------------------------------------- 1.4/1.4 MB 4.8 MB/s eta 0:00:00
Downloading numpy-1.26.4-cp312-cp312-win_amd64.whl (15.5 MB)
   ---------------------------------------- 15.5/15.5 MB 5.9 MB/s eta 0:00:00
Installing collected packages: numpy, shapely
  WARNING: The script f2py.exe is installed in 'C:\Users\...\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed numpy-1.26.4 shapely-2.0.3

This is a common error when python is used by a layman like me...

Rickthebig, 2024-03-21:
The script won't run:
python3.12.exe softfever_slicer_post_processing_script.py
Traceback (most recent call last):
  File "...\softfever_slicer_post_processing_script.py", line 628, in <module>
    class BridgeInfill():
  File "...\softfever_slicer_post_processing_script.py", line 629, in BridgeInfill
    def __init__(self,pts=[],id=random.randint(1,1e10)) -> None:
                                ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.752.0_x64__qbz5n2kfra8p0\Lib\random.py", line 336, in randint
    return self.randrange(a, b+1)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.752.0_x64__qbz5n2kfra8p0\Lib\random.py", line 312, in randrange
    istop = _index(stop)
            ^^^^^^^^^^^^
TypeError: 'float' object cannot be interpreted as an integer

Although I installed python 3.9, the libraries are found only when using python 3.12

I will have to run the script using python 3.19 and see.

Rickthebig, 2024-03-21:
I got the versions straight, now processing the script using python 3.9 I get:
File "softfever_slicer_post_processing_script.py", line 325, in <module>
    def getPtfromCmd(line:str)->Point | None:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'

Perhaps I will have to dive into python to at least correct the syntax errors!

scrampker, 2024-04-02:
@Rickthebig 

Use WSL Ubuntu if you're wanting to run Python in a linux environment.

closed 2024-07-10 by github-actions[bot]

federicoparra, 2024-08-21:
> Working on implementing this on my fork: https://github.com/scottmudge/OrcaSlicer/tree/feature_arc-overhangs
> 
> No promises yet -- there are significant differences between OrcaSlicer and SuperPleccer, which may require some debugging/modifications.
> 
> If I get something working, I'll submit a PR

Hi! have you gotten to develop a working version of the algorithm in your branch? Should I compile it and try it? I use a bambu lab A1 and with it I have no luck with the post-processing script :(

boromyr, 2024-09-06:
+1 for this function, the postprocessor in Py is discreet but not exceptional

Anthony-Bec, 2025-01-03:
> Hi! have you gotten to develop a working version of the algorithm in your branch? Should I compile it and try it? I use a bambu lab A1 and with it I have no luck with the post-processing script :(

same for me

fernplant, 2025-03-15:
I would like to see this as well

cvanelteren, 2025-08-01:
why is this bot closing this issue as not planned without any hands on from the devs?