OrcaSlicer/OrcaSlicer #903: [Feature Request] GCode Substitutions or Prevent ozzing after M600 (Non AMS Printer)

fontenele, 2023-04-30:
**Is your feature request related to a problem? Please describe.**
When using not AMS printer and we have to use M600 to change filament, to prevent oozing in prusaslicer we have to comment next line after M600 using gcode substitutions. [Reference](https://github.com/prusa3d/PrusaSlicer/issues/8047#issuecomment-1413816569)

**Describe the solution you'd like**
I dont know if Orca have an option for this issue, if already have sorry my mistake here.

**Describe alternatives you've considered**
Prevent negative retraction after M600.

**Additional context**
Some print about workaround this problem
[image]
[image]
[image]

SoftFever, 2023-05-01:
Orca does do extra retraction for filament change.
The setting is, however read-only in current versions.
Gonna change that
[image]
[image]

SoftFever, 2023-05-21:
Fixed in 1.6.3 beta

closed 2023-05-21 by SoftFever