OrcaSlicer/OrcaSlicer #905: Remove the limit for layer height

julie777, 2023-04-30:
**Describe the bug**

P1P using .4mm nozzle and Bambu Slicer will not let me set layer height to .3 which is very common for a .4 nozzle. Instead it limits the layer height to .28.

[image]

Describe the solution you'd like

Allow me to set the layer height that I choose. The print profiles provides show me what BL thinks is a good setting. I get that. However, there are many reasons to choose a different height.

**Desktop (please complete the following information):**
 - OS: Windows]
 - Version 1.6.2

TechDude23, 2023-05-01:
Hi Julie777

You can change it in the Printer Settings go to the extruder tab and layer height limits - Max 0.28 and change it to 0.32 and save settings

derx05, 2023-05-02:
Hey guys,
quick note. This issue can cause an endless  instant crash loop. When I open my slicer I get exactly that error and it instantly crashes. (Version 1.6.2)
Sadly I don't know which config file includes this value so I am stuck!

SoftFever, 2023-05-02:
> Hey guys, quick note. This issue can cause an endless instant crash loop. When I open my slicer I get exactly that error and it instantly crashes. (Version 1.6.2, MacOS) Sadly I don't know which config file includes this value so I am stuck!

Is there any useful info in the log file?

derx05, 2023-05-02:
> > Hey guys, quick note. This issue can cause an endless instant crash loop. When I open my slicer I get exactly that error and it instantly crashes. (Version 1.6.2, MacOS) Sadly I don't know which config file includes this value so I am stuck!
> 
> 
> 
> Is there any useful info in the log file?

Will post as soon as I am back home

derx05, 2023-05-02:
[Crash Logs.zip](https://github.com/SoftFever/OrcaSlicer/files/11375961/Crash.Logs.zip)
 Here are the 2 logs. The one Orca Drops in its logs folder and the one macOS generates. Honestly I didn't spent much time looking into them, cause I am not into this project. Hope it helps

hliebscher, 2023-05-02:
@derx05  i think softfever needs the log files from /Users/?????/Library/Application Support/OrcaSlicer/log/

derx05, 2023-05-03:
> @derx05 i think softfever needs the log files from /Users/?????/Library/Application Support/OrcaSlicer/log/

One of the logs provided in the zip is exactly that log file. But I can't find an error in them so either their logging is not working or that crash is not causing a log entry.

SoftFever, 2023-05-04:
> [Crash Logs.zip](https://github.com/SoftFever/OrcaSlicer/files/11375961/Crash.Logs.zip) Here are the 2 logs. The one Orca Drops in its logs folder and the one macOS generates. Honestly I didn't spent much time looking into them, cause I am not into this project. Hope it helps

I can't find useful info in the log file either.

Anyway, you can manually add the following line in your custom profile(.json). It may fix the issue for you.
I suspect the value in your custom profile is corrupted somehow.
```
    "max_layer_height": [
        "0.3"
    ],
```

derx05, 2023-05-04:
> I can't find useful info in the log file either.
>
> Anyway, you can manually add the following line in your custom profile(.json). It may fix the issue for you. I suspect the value in your custom [...]

Yeah I couldn't find the paramenter I solved it yesterday by deleting the user folder.

SoftFever, 2023-05-06:
Closing as problem solved.

closed 2023-05-06 by SoftFever