OrcaSlicer/OrcaSlicer #906: reprap firmware (duet wifi)

broncosis, 2023-04-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

broncosis, 2023-04-30:
here is a well built Prusa profile for a railcore 
https://github.com/JohnOCFII/RailCore/tree/master/slicers/PrusaSlicer
would be awesome if I could use the same slicer for both machines

hliebscher, 2023-05-01:
pls describe  what you want to happen.
duet wifi runs like it is ... you can set your custom printer and under settings HOST TYPE = DUET

broncosis, 2023-05-05:
Mostly I think after poking around a bit it does support it 
just doesn't have a option from the printer setup menu 

so you have to choose a Marlin printer then edit it from there 
to use Reprap firmware   

its a small thing I know if its a hassle I get it 
but the point of a feature request is to ask and provide Ideas

mydevpeeps, 2023-06-30:
in cura the duet plug does the exact same thing the x1c does.  send or print (send and print from card). like to see this available for duet as well

broncosis, 2023-06-30:
This is available in superslicer and prusa aswell would be a decent addon.

closed 2023-11-21 by github-actions[bot]