OrcaSlicer/OrcaSlicer #909: Print multiple layers per material to massively reduce material changes in multi-material prints.

ewoolsey, 2023-05-01:
When printing with the AMS, frequent material changes which occur every layer lead to extremely long print times and significant material waste. Similar to the `print sequence` feature, it would be possible on many projects to print multiple layers at a time of each material sequentially. Even a simplistic implementation which prints 2-5 layers at a time and performs a z-hop in between parts could eliminate 50-80% of material changes. A more advanced implementation which takes into account the toolhead geometry could perform even better.

IMO this feature would be absolutely game changing, and I don't believe implementation would be that difficult (Though I have no experience developing OrcaSlicer). I would love for someone with experience to comment on the feasibility of this.

SoftFever, 2023-05-02:
Some thoughts here:
1. The nozzle traveling algo needs to be updated to avoid colliding with the printed part.
2. This is probably the trickiest part. Imagine this situation, the nozzle collides with the printed part depending on the height difference and nozzle shape. These kinds of limitations make this approach much less effective than expected.
[image]

rizwansarwar, 2023-10-04:
I think it is a great idea. For majority of the prints this could be dealt with using zhop i-e have higher zhop where multiple layers are done to save color changes. If implemented, it should be a configurable option to choose the number of layers to combine.

closed 2024-01-10 by github-actions[bot]