OrcaSlicer/OrcaSlicer #920: Linux - Device tab not displaying klipper interface

TexZeTech, 2023-05-01:
**Describe the bug**
The device tab is not displaying the mainsail interface.

**To Reproduce**
1.) Install a fresh instance of Manjaro in a vm (as of this time [manjaro-kde-22.1.0-230421-linux61.iso](https://download.manjaro.org/kde/22.1.0/manjaro-kde-22.1.0-230421-linux61.iso) )
2.) Install Appimagelauncher and dependencies 
3.) Install webkit2gtk, Webkit2gtk-4.1, Webkit2gtk-5.0 and any dependencies
4.) Run appimage (integrate or just run it)
5.) Add a generic klipper printer and install Bambu network 
5.5) try the device tab to see if the video plays.
6.) Go to network on the printer and enter IP address -> enter -> test
7.) Close the network config and click on "Device" 
8.) nothing

**Expected behavior**
To use the device tab to interact with Mainsail (or other *not tested*) interface(s).

**Screenshots**
[image]

**OS & Hardware of main machine:**
Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.4
KDE Frameworks Version: 5.104.0
Qt Version: 5.15.8
Kernel Version: 6.1.25-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2

**OS & Hardware of VM:**
Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.4
KDE Frameworks Version: 5.104.0
Qt Version: 5.15.8
Kernel Version: 6.1.25-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 5.7 GiB of RAM
Graphics Processor: llvmpipe
Manufacturer: innotek GmbH
Product Name: VirtualBox
System Version: 1.2

**Aditional notes:**
This was working at some point previously on my system. :shrug: 
Any help would be appreciated.

hliebscher, 2023-05-02:
can you open the web interface in firefox?

hliebscher, 2023-05-02:
sometimes have other private ip networks no routing  pls test in VM

H4kunaM4tata, 2023-05-03:
This is actually awesome, I didn't know this was a thing.

Mint Cinnamon (DEB) is working fine.

[image]

@TexZeTech if you don't mind me asking why VM and why Manjaro instead of a popular and stable distro like Mint?? Or even worse, Ubuntu???
I remember trying it with Windows VM when Linux did not even had AppImage and boy, what a poop.

Either way, make sure the VM network isn't set as NAT but Bridge without replicating, so the VM can have a physical network access without virtualization. NAT can be a poop sometimes.

TexZeTech, 2023-05-03:
@H4kunaM4tata Happy you discovered a feature :+1: 

As to your questions the VM was to verify it's not something I did with my main computer as for why Manjaro & not Ubuntu, Debian, Fedora, Windows, Mac... preference at this point in time?

I'm just trying to find out what is missing as to end up breaking this function, as it was working at a previous point in time. I have installed what I perceive as the dependencies required with no change in that so it's going to be something dumb.

And I just tried v1.5.0 and had the same results.

I will go poke the Manjaro forum as well when I get a chance & leave this open so when I get results I can post the fix here as well.

H4kunaM4tata, 2023-05-03:
I see, for what it's worth it, when we started supporting the guys to have the AppImage, It was a total rollercoaster and ironically SoftFever back then was the only version giving way less headache.

Ticket number is #12 I think under BambuLab repository, you might ask there, it will be a lot easier and faster to have a feedback since its still active and being supported by legends.

adrian-dobre, 2023-05-07:
This started happening to me on Mac OS (Intel) today, after installing the latest version. Switched back to the old one (was still installed) and the problem remains.

bjw-s, 2023-06-20:
Chiming in with a +1 on MacOS. On the very first startup I can see the howto video. I went on to set up the physical printer, closed and reopened Orca and then I am just getting a white screen whenever I go to the Device tab

Rokin05, 2023-06-20:
Same bug with OrcaSlicer version 1.6.3 on Arch (Aur: https://aur.archlinux.org/packages/orca-slicer)

TexZeTech, 2023-08-03:
Still present in Orcaslicer V1.6.4 Beta app image. I'll install a fresh vm of Manjaro later and see if anything changes in that.

Jakeo-93, 2023-08-04:
+1 for this. Appimage on a arch based distro and the device tab remains black. Mainsail works fine in webbrowser

Spotigus, 2023-08-07:
+1 as well. Does not work on linux mint.

TexZeTech, 2023-08-24:
Still present in 1.6.6 appimage

```

Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.7
KDE Frameworks Version: 5.108.0
Qt Version: 5.15.10
Kernel Version: 6.1.44-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2
```

TexZeTech, 2023-09-12:
Still present in 1.7.0 beta appimage.

R4toLoko, 2023-09-12:
It stopped working for me after 1.6.6
For what it is worth, there seems to be some weird network stuff going on.
The slicer somewhat is still sending files over to the printer, Voron, so I am okay with that.

djovic78, 2023-09-12:
Still present with the 1.7.0 beta appimage on KDE Neon (Basically Ubuntu)

Operating System: KDE neon 5.27
KDE-Plasma-Version: 5.27.7
KDE-Frameworks-Version: 5.109.0
Qt-Version: 5.15.10
Kernel-Version: 5.17.0-1035-oem (64-bit)
Grafik-Plattform: Wayland
Prozessoren: 8 × 11th Gen Intel® Core™ i7-1165G7 @ 2.80GHz
Speicher: 31,1 GiB Arbeitsspeicher
Grafikprozessor: Mesa Intel® Xe Graphics
Hersteller: LENOVO
Produktname: 20Y0S00200
Systemversion: ThinkPad X1 Yoga Gen 6

Dev01-D, 2023-09-15:
Confirmed issue with latest 1.7 appimage, I get this with my Linux box running POP!_OS
Device connection confirms test is ok.

Using normal web browser works fine.

alanmakestuff, 2023-09-17:
> Confirmed issue with latest 1.7 appimage, I get this with my Linux box running POP!_OS Device connection confirms test is ok.
> 
> Using normal web browser works fine.

Same with my Pop_OS 22.04 LTS. Blank tab. Works fine in browser. Installed Appimage.

Cyber-Garret, 2023-09-26:
Same for me on Linux mint 21.1 cinnamon edition, Fluidd work in device tab, but "Mainsail" show blank tab = (
P.S. in Chrome browser all work as expected.

eloo, 2023-09-26:
i have tested some stuff with orca-slicer and mainsail but sadly not found the cause..

but maybe it helps for further investigation.

i have tested the following:

- Disable cors in moonraker -> no change
- Disable authorized clients in moonraker -> no change
- Check logs from mainsail (nginx) -> call to mainsail looks okay, at least requests are send and response code looks good
- Append some path to url (like webcam) -> works, but its the landing page of webcam
- Append "index.html" to url -> does not work
- "Deactivate" the websocket -> does not work

so i guess the issue is "somewhere" loading and processing the javascript stuff.. 

but, i have seen that bambustudio now supports third party printers as well.. so i will give this a try and maybe @SoftFever can find the "missing piece" here
https://github.com/bambulab/BambuStudio/commit/efd65561a2de060c154eb5e7fb559bb7a8ae0a17

funfact: bambustudio webview is based on the code from here ;) 

i will update when i have tested bambustudio

eloo, 2023-10-06:
i'm not sure what happend.. 

but the device tab is now working for me :D 

here are my current versions in mainsail
[image]

the "Device UI" is set a DNS name in local network (http://mainsail.lan) 

orcaslicer version: 1.7.0
os: manjaro

TexZeTech, 2023-10-08:
Well, my laptop seems to work again after a system update, my tower on the other hand is more broken now.... I'll need to find out what the difference is between the packages installed on the systems.

urgggg

ReneSchmidts, 2023-10-10:
Here some news that might be helpfull searching for the issues source (I hope).
I do have a second printer running a DUET2 with RepRap. This was working correctly in the Orca device tap until a few minutes ago. Linux Mint installed an update to GTK & Java...... since that time the Orca device tap is staying white for Klipper and RepRap
[image]

darco415, 2023-10-18:
Im on Linux Mint, I switched over to Nouveua drivers and the black screen issue went away, Im back because now I get a :Could not connect: Connection refused error when opening the device tab. Browser works fine...using 1.8 dev but 1.7 is also a problem

rvanderson, 2023-11-03:
Not sure if this is the same problem anyone else is having, but I had both a username and an API in the connection screen. Removing the name and using only the API key seemed to fix the problem immediately.

darco415, 2023-11-19:
> Not sure if this is the same problem anyone else is having, but I had both a username and an API in the connection screen. Removing the name and using only the API key seemed to fix the problem immediately.

wow dude, that fixed it for me!..running 1.8 stable

adrian-dobre, 2023-11-25:
Clearing the Device UI field fixed it for me also:
[image]

TexZeTech, 2023-11-27:
> Clearing the Device UI field fixed it for me also: ![Screenshot 2023-11-25 at 16 29 40](https://private-user-images.githubusercontent.com/50677374/285597155-c3a3bb54-e4fa-498e-a490-f266235a8b60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExMDQzMTMsIm5iZiI6MTcwMTEwNDAxMywicGF0aCI6Ii81MDY3NzM3NC8yODU1OTcxNTUtYzNhM2JiNTQtZTRmYS00OThlLWE0OTAtZjI2NjIzNWE4YjYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI3VDE2NTMzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIwYmNjMWQzNDg5NmRkYmM1OWQ2ZWUwOTk2YzYwZTAyOWU3Njg5M2NhMTdmYWE4MDEzYWVlMmM1OGMwZGI4NTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.mZvAtpQYsqsGjN5gnhP1Hw9M9_1mvZpB46HeKt7p5A8)

I don't even know anymore it's working now maybe something was in there for me.

closed 2023-11-27 by TexZeTech

TexZeTech, 2024-03-28:
So I think I just found the fix for my issue. 

I'm running NVIDIA proprietary drivers.
I added `nvidia-drm.modeset=1 nvidia-drm.fbdev=1` to my Grub boot and that seems to have fixed it.
I'm also currently on X11 for my Graphics platform.

Akucuki, 2024-04-12:
> So I think I just found the fix for my issue.
> 
> I'm running NVIDIA proprietary drivers. I added `nvidia-drm.modeset=1 nvidia-drm.fbdev=1` to my Grub boot and that seems to have fixed it. I'm also currently on X11 for my Graphics platform.

The same issue, and enabling drm didn't help, having the same problem with the home and project tabs and the app in general is very slow to switch between any of the tabs.

Akucuki, 2024-04-12:
The problem seems to be with the webkit2gtk with the versions higher than 2.42.5-2. Downgrading back to 2.24.5-2 eliminates all of the issues with blank and laggy tabs. Works like charm now.

Zincr0, 2024-06-21:
Don't know why this bug is "closed", the issue still happens on orca 2.1.0

TexZeTech, 2024-06-22:
> Don't know why this bug is "closed", the issue still happens on orca 2.1.0

Because I've experienced and continue to experience this + new/different issues on my old computer that has an old install that is likely broken in some dumb way and I cannot replicate this on my laptop as stated above. 

So the issue seems to be related to [libwebkit2gtk](https://github.com/SoftFever/OrcaSlicer/issues?q=libwebkit2gtk-4.0.so.37) from what I've seen.

I also did "fix" the issue at the time of me closing this by adding the nvidia.drm tag in my grub file. I have since had the same issue pop up again and my welcome screen is blank but not on my laptop.

You also haven't added anything useful to try and debug this like below (at a minimum). 

I would also recommend adding what Graphic drivers you are using Nvidia with nouveau or propriatry, etc. + log files.

```
Operating System: Manjaro Linux 
KDE Plasma Version: 6.0.5
KDE Frameworks Version: 6.2.0
Qt Version: 6.7.1
Kernel Version: 6.6.32-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i5-3337U CPU @ 1.80GHz
Memory: 3.3 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 4000
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: TAICHI31
System Version: 1.0
```