OrcaSlicer/OrcaSlicer #924: Orca prints way too big.

Rigattoni, 2023-05-02:
**Describe the bug**
After printing the attached 3mf I found out that the print was nearly 3mm too large in x and y.
[image]

The print came out with size 82,9x 82,9 so completely oversized.
I checked the gcode file from Orca in gcode viewer online:
[image]

The same 3mf was printed exactly in the correct measures when I used SuperSlicer to print this part.

Hint: It seems to be only this single part which is oversized. I sliced and printed a test part of 100x100x3mm to verify the correct calibration.

**3mf File for This Bug**
[Salzmuehle.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/11376016/Salzmuehle.3mf.zip)

**To Reproduce**
- Import the attached 3mf into Orca and slice it.
- print this file or check in gcode viewer online

**Expected behavior**
I just want to get a correct sized object in print.

**Printer model**
RatRig V-Core 3

**Desktop (please complete the following information):**
 - OS: Windows]
 - Version 11

Rigattoni, 2023-05-02:
Retested this file with OrcaSlicer V1.6.2 Offical Release.
Same oversize visible.

hliebscher, 2023-05-02:
i have print your model it woks ! no oversize 
but wrong values in g-code analyser
but exact 80mm in print

[image]
[image]
[image]

Rigattoni, 2023-05-02:
I printed it twice completely and always got the same dimensions as also shown in the gcode analyzer.
[image]

hliebscher, 2023-05-02:
@Rigattoni  habs gefunden... schreibe mal gleich auf deutsch;-) bei dem Filament ist 96% schrumpfen angegeben 
und die 4% die sind genau die Differenz!    ich glaube da muss 99,6 rein  ABS schrumpft doch nicht 4% oder???

Rigattoni, 2023-05-03:
@hliebscher 
Du hast Recht. Genau in dem einen Profil (warum auch immer) stehen 96% Schrumpfung drin.
Eigentlich steht mein eSun ABS+ sogar auf 100% in allen anderen Profilen, weil bis auf ein paar 100tel hat das in der Kalibrierung gepassst.
Wie das da rein gekommen ist... keinen blassen Schimmer. 99,6% wird bei mir sogar bei kleinen Objekten etwas zu klein.

Vielen Dank für deine Mühe. Fehler gefunden, gerade geändert in dem Profil. Alle anderen Profile stehen dummerweise richtig.

Fehler 60 also. :-)

closed 2023-05-03 by Rigattoni