OrcaSlicer/OrcaSlicer #930: Max flowrate calibration in 1.6.3 beta

RaySajuuk, 2023-05-03:
**Describe the bug**
The max flowrate calibration in v1.6.3 generates a very tall model that takes a few hours to finish printing. I would assume the generated model should be similar dimensions in v1.6.2.

**3mf File for This Bug**
Built-in max flowrate model

**To Reproduce**
Generate max flowrate calibration via the calibration menu in OrcaSlicer any version of v1.6.3.

**Expected behavior**
I would assume the calibration model should be similar to the previous model.

**Screenshots**
[image]

**Printer model**
BBL X1CC

**Desktop (please complete the following information):**
 - OS: [MacOS]
 - Version [Ventura 13.4]

psiberfunk, 2023-05-05:
There's a 1.6.3 beta somewhere?  are we talking about nightlies?

RaySajuuk, 2023-05-05:
Happy to report the issue is resolved.

closed 2023-05-05 by RaySajuuk

SoftFever, 2023-05-05:
> There's a 1.6.3 beta somewhere? are we talking about nightlies?

is referring to the nightly build