OrcaSlicer/OrcaSlicer #936: Help menu disappears after few seconds

RaySajuuk, 2023-05-03:
**Describe the bug**
The help menu disappears after a few seconds.

**To Reproduce**
Launch Orcaslicer help menu will disappear soon after.

**Expected behavior**
The help menu should remain in the menu bar.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: [macOS]
 - Version [13]

RaySajuuk, 2023-05-06:
I am running it on  Macbook Pro notched display, and the menu bar has enough free space to show all items.

liftbag, 2023-05-13:
I have the same issue.

macos 13.3.1 (a) apple silicon
OrcaSlicer Build Mac AppleSilicon 319: Commit (951293c) pushed by SoftFever

https://github.com/SoftFever/OrcaSlicer/assets/14921654/fb3223d4-577d-448c-a84e-31f5f62cb491

RaySajuuk, 2023-11-01:
Issue resolved

closed 2023-11-01 by RaySajuuk