OrcaSlicer/OrcaSlicer #946: External Spool feature missing from 1.6.2?

Photogad, 2023-05-04:
**Describe the bug**
Not sure if it's a bug or user error but external spool support seems to be missing from OrcaSlicer 1.6.2 (is it not built against Bambu Studio 1.6.2?? They introduced external spool support in 1.6.2)

**Screenshots**
[image]

**Printer model**
Bambu X1CC

**Desktop (please complete the following information):**
Windows
1.6.2 latest public release as of 5/2/22

fr3ak2402, 2023-05-04:
This is not a bug, but the feature is not yet included in version V1.6.2. This is currently only included in the nightly build of V1.6.3 and then also in the official V1.6.3 of OrcaSlicer.

Photogad, 2023-05-09:
> This is not a bug, but the feature is not yet included in version V1.6.2. This is currently only included in the nightly build of V1.6.3 and then also in the official V1.6.3 of OrcaSlicer.

Good to know.
I suggest kindly to @SoftFever to keep naming convention of OrcaSlicer versions to match closer to Bambu Studio. I realize OrcaSlicer is being developed to be its own slicer software, but if it's built against Bambu Studio it can help avoid confusion.

fr3ak2402, 2023-05-09:
@Photogad 
I have proposed this adjustment. You can take a look at the proposal.
Improvement of the display of the basic version of Bambu Lab Studio in OrcaSlicer and in Github. #961

Photogad, 2023-05-14:
> @Photogad I have proposed this adjustment. You can take a look at the proposal. Improvement of the display of the basic version of Bambu Lab Studio in OrcaSlicer and in Github. #961
@fr3ak2402 
thank you

Photogad, 2023-05-14:
I am closing this for now. Question has been answered.

closed 2023-05-14 by Photogad