OrcaSlicer/OrcaSlicer #948: [Voron] Send to print is cutting out the info

H4kunaM4tata, 2023-05-04:
First of all thank you, this is freaking awesome, I never used this with X1C because it requires cloud.
With my Voron this works flawlessly, THANK YOU!

I did notice tho the info which appears to inform that if you wanna change the folder level, you must use slashes, isn't being displayed properly.
Nothing wrong with the feature, just "cosmetics"

Thank you

[image]
[image]

hliebscher, 2023-05-04:
[image]

For me it works fine...
MAC M1 1.6.3beta

H4kunaM4tata, 2023-05-05:
> [image]
> 
> For me it works fine...
> MAC M1 1.6.3beta

Weird, it must be a Linux thing then.
For what matter the feature is working flawlessly.
Thanks for sharing that.

R4toLoko, 2023-08-16:
This is still a cosmetic issue on Linux @SoftFever 
Not a biggie, can be closed.

[image]

R4toLoko, 2023-09-12:
This is just a cosmetic thin, it isn't fixed and other parts have the same but I am closing it.

closed 2023-09-12 by R4toLoko