OrcaSlicer/OrcaSlicer #969: At first Setup wizard, List is empty in printer selection part

sagesin, 2023-05-06:
[image]
It's my first use of Orca Slicer, and I downloaded the v1.6.2 and can't do anything in printer selection part.

It doesn't show any Printer and I can't skip it. 

I'm using Windows 11

Did i do something wrong? or how can i fix it?

Ergonomicmike, 2023-07-05:
I tried Orca Slicer for the first time, v1.6.3. On start up, I too, have a blank Setup Wizard, Except that my popup window is TOTALLY blank - no text nor buttons. Win10 64-bit, but stripped down using NTLite.

Ergonomicmike, 2023-07-06:
More data: When I try Orca Slicer in a VM running a stripped down version of Win10, I receive a error saying "MSVCP140.dll not found." After acknowledging that error, then "VCRUNTIME140.dll was not found." Finally, "VCRUNTIME140_1.dll was not found."

So it appears that Orca Slicer needs a Visual C++ runtime.

I'll try it in a vanilla Win10 in a VM and report.

Ergonomicmike, 2023-07-06:
More: I had to install/update vcredist2015_2017_2019_2022_x64.exe (from Administrator) in Windows 10 (on my VM) for orca slicer to run. It ran. But then a warning about my video driver not being good enough.

So I changed the video driver in VBox to allow 3D. (Open GL.) Once done, orca slicer works on a vanilla Win10 in my VM.

Now, on my stripped down (ntlite'd) Win10 desktop, the Setup Wizard still does not populate. So apparently I have removed something from Windows that orca slicer needs for the Setup Wizard to operate correctly.

closed 2023-11-21 by github-actions[bot]

burak1000, 2024-03-25:
Soluiton is very simple:  use Ultimaker Cura.

fabiovix, 2024-12-03:
I do have this problem too (Ubuntu 24)

burak1000, 2024-12-03:
> I do have this problem too (Ubuntu 24)

Soluiton is very simple: use Ultimaker Cura.