OrcaSlicer/OrcaSlicer #979: Wipe moves happen outside of their object tags

kellyinvestor, 2023-05-06:
**Describe the bug**
Orcaslicer produces retraction moves (wipe) that are not within an exclude object tag.

Orcaslicer
```
G1 X161.305 Y190.04 E.03405
G1 X160.907 Y190.497 E.02568
; stop printing object OrcaCube id:1 copy 0
EXCLUDE_OBJECT_END NAME=OrcaCube_id_1_copy_0
;LAYER_CHANGE
;Z:0.6
;HEIGHT:0.2
;WIPE_START
G1 F5872.472
G1 X161.305 Y190.04 E-.17275
G1 X161.305 Y190.434 E-.11225
;WIPE_END
G1 E-.015 F1800
SET_PRINT_STATS_INFO CURRENT_LAYER=3
; printing object OrcaPlug id:0 copy 0
EXCLUDE_OBJECT_START NAME=OrcaPlug_id_0_copy_0
G1 X149.8 Y107.44 Z.6 F15000
G1 E.3 F1800
M204 S1000
;TYPE:Inner wall
;WIDTH:0.449999
```

Superslicer
```
;LAYER_CHANGE
;Z:0.88
;HEIGHT:0.28
;WIPE_START
G1 F6240
G1 X157.958 Y128.083 E-0.285
;WIPE_END
G1 E-0.015 F2400
G1 Z1.2 F7800
; stop printing object Shape-Cylinder id:1 copy 0
EXCLUDE_OBJECT_END NAME=Shape_Cylinder_id_1_copy_0
; custom gcode: layer_gcode
SET_PRINT_STATS_INFO CURRENT_LAYER=3
; custom gcode end: layer_gcode

G92 E0
G1 Z1.48
; printing object Shape-Box id:0 copy 0
EXCLUDE_OBJECT_START NAME=Shape_Box_id_0_copy_0
; acceleration to travel
M204 S2000
G1 X161.937 Y165.305
```

**To Reproduce**
Slice any file with two or more objects an observe wipe moves that are not where they should be

**Expected behavior**
Wipe move should be included in the previous object.

**Desktop (please complete the following information):**
 - OS:  Windows 11
 - 1.6.2 and dev

bordermultimedia, 2023-09-16:
I came to report the same bug, when printing sequentially ("by object"instead of "by layer") and using the exclude object, the printer strill executes several movements, including retractions that later will cause the next object print to fail:

This is how it looks with label_object turned on but exclude_object turned off, and pre-processed by Klipper itself:

`G1 X143.053 Y64.648 E.07099
; stop printing object Cuerpo B.stl id:1 copy 0
EXCLUDE_OBJECT_END NAME=Cuerpo_B_stl_id_1_copy_0
M106 S219
;LAYER_CHANGE
;Z:0.36
;HEIGHT:0.16
;BEFORE_LAYER_CHANGE
;0.36
TIMELAPSE_TAKE_FRAME
G92 E0

;_SET_FAN_SPEED_CHANGING_LAYER
; printing object Cuerpo B.stl id:1 copy 0
EXCLUDE_OBJECT_START NAME=Cuerpo_B_stl_id_1_copy_0
G1 E-3.15 F2400
;WIPE_START
G1 F1800
G1 X142.346 Y63.941 E-1.35`

This is what OrcaSlicer generates when exclude_object is turned on, ad Klipper skips pre-processing the file:

`G1 X143.053 Y64.648 E.07099
; stop printing object Cuerpo B.stl id:1 copy 0
M106 S255
;LAYER_CHANGE
;Z:0.36
;HEIGHT:0.16
;BEFORE_LAYER_CHANGE
;0.36
TIMELAPSE_TAKE_FRAME
G92 E0

EXCLUDE_OBJECT_END NAME=Cuerpo_B.stl_id_1_copy_0
;_SET_FAN_SPEED_CHANGING_LAYER
; printing object Cuerpo B.stl id:1 copy 0
G1 E-3.15 F3000
;WIPE_START
G1 F1800
G1 X142.346 Y63.941 E-1.35
;WIPE_END
EXCLUDE_OBJECT_START NAME=Cuerpo_B.stl_id_1_copy_0
G1 X141.546 Y56.35 Z.6 F9000
G1 X140.88 Y50.018 Z.6`

It seems bot the EXCLUDE_OBJECT_START NAME and EXCLUDE_OBJECT_END NAME are both off, too far from the corresponding object label.

bordermultimedia, 2023-12-18:
last time I checked the bug still exists, and it's a big one, If I ever forget to untick "exclude objects" when I use the printing order "by object", too many movements and retractions happen when they should have been skipped, and the following object print fails completely.

bordermultimedia, 2024-01-13:
Bambu studio version 1.8.2.56 does not have this issue (on an Ender 3), hopefully that part of the code will be ported in next version (1.9 version still has this issue)

closed 2024-04-20 by github-actions[bot]

bordermultimedia, 2024-05-04:
The issue is still there, retraction and wipe movements happen outside the object tags, and that makes the "exclude objects" unusable when printing sequence is set to "by object" (because a retraction will be executed for every layer of the excluded object, and the next object to be printed will certainly fail). Bambu Studio 1.8.4.51 (latest version) doesn't seem to have this problem.

johyphenel, 2025-06-05:
Perhaps related — it seems like autogenerated purge towers also generate a wipe way outside the bounds of the purge tower object on the first layer — I only noticed because this out outside my print area and killed the print the first time. 

[image]