OrcaSlicer/OrcaSlicer #980: Presets return to old settings.

KaiserCorp, 2023-05-06:
On Orca-slicer 1.6.2-beta, after saving Filament and print settings and restarting Orca-slicer, old presets return.
some presets will be changed but not all presets,
Also deleting presets brings back the presets after restarting Orca-slicer.

I expect the settings to be saved correctly and deleted when i want to delete settings.

 - OS: Microsoft windows
 - Version: Windows 10 pro x64

ghubsch, 2023-05-30:
Same issue here. Lost hours of tweaking profiles :(

cactixxx, 2023-06-01:
Had the same issue on Orca-slicer 1.6.3 Linux.  I fixed it by enabling "Auto Sync User Presets" under preferences.

bjerabek, 2023-06-13:
I faced similar issue after upgading to 1.6.3 from 1.6.2, my earlier saved presets are not visible in the newer version.
After some searching I found that the presets are stored in JSON files under the path: c:\Users\<username>\AppData\Roaming\OrcaSlicer\user\default\

After creating new customized preset with 1.6.3 version I realized that the difference between the old and new one was the *inherits* field. In 1.6.2 it was pointing to 'Generic PLA @Creality' while in the new one it is 'Creality Generic PLA'.
After correcting the inherits field in all JSON files the presets became visible in 1.6.3.

closed 2023-11-21 by github-actions[bot]