OrcaSlicer/OrcaSlicer #982: Typo in initial Setup Wizard

MikeOxlong83, 2023-05-07:
**Describe the bug**
[image]

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Open the program for the 1st time, or start the Setup wizard manually by adding a printer.

**Expected behavior**
An interface free of typos.

**Screenshots**
See above

**Printer model**
All

**Desktop (please complete the following information):**
 - OS: Mac Intel
 - Version Latest (1.6.2)

hliebscher, 2023-05-07:
is fixed with my other PR

hliebscher, 2023-05-09:
@SoftFever @MikeOxlong83  pls close the bug

closed 2023-05-10 by SoftFever