OrcaSlicer/OrcaSlicer #983: Multiple Device Tabs

MiscPrinter, 2023-05-07:
**Is your feature request related to a problem? Please describe.**
I use Orca to prepare prints and send them to multiple machine on BL and Octoprint. Going to prepare, choosing the machine, then switching back to the device tab is slow when monitoring multiple print happening at the same time. 

**Describe the solution you'd like**
Having multiple "Device" tabs that allow the user to select which printer on which tab to monitor and at the same time relabel that "Device" tab name to the name of the printer for ease of use. Having the number of device tabs be user adjustable to suit each person needs. The options should allow for choosing BL and non-BL printer to be viewed in separate tabs at the same time. 

**Describe alternatives you've considered**
Allowing multiple Orca app instances to be open at the same time. This may be OS limited or app limited. I'm not sure but it is an alternative I thought. Having printer A on one app instance and printer B on the other instance. 

**Additional context**
Awesome job making this software. Thanks.

kellyinvestor, 2023-05-07:
I'd like this even for use with a single machine, for the mainsail interface in one tab and prettygcode view in the other.

hliebscher, 2023-05-08:
you can open under file new Window a second instance for the same printer or other printers with other interfaces
i use 3 different printer so i have 3 different windows open

MiscPrinter, 2023-05-08:
What OS are you using that's allowing you to open multiple instances of the app? I'm running MacOS and don't see an option to have multiple instances open.

julie777, 2023-05-08:
I have a different idea for this. I would like the device tab to show the list of know devices on the left sidebar. Move all the stuff from the tabs on the left onto the single page for that device. 

It is ridiculous to have to go and look to see if there is an HMS event, or if there is a firmware update available.

The sidebar list of printers should show a simple status (as part of the icon, or another way). That way you have a list of all known printers. 
You can see which printers are offline.
You can see which printers are currently printing.
You can see which printers have finished a print and need the plate cleared to make them ready.

You can probably tell that I think the device tab should evolve into the place to manage printers, print queuing, and all the other functions that one needs with multiple printers/a print farm.

In addition to moving the info from the tabs on the left, the printer status pane should also list any prints that have been queued for that printer specifically. (I am assuming that full featured printer management would include queuing up prints in a way that the first printer meeting the requirements for the print would be used.

hliebscher, 2023-05-09:
@MiscPrinter open the app once then go to top menu  File -> New Window  than open a second window with the app

at MAC OS

[image]

MiscPrinter, 2023-05-09:
@hliebscher Awesome. Thanks for pointing that out.

PrintMan133, 2024-01-11:
> Tôi có một ý tưởng khác cho việc này. Tôi muốn tab thiết bị hiển thị danh sách các thiết bị đã biết ở thanh bên trái. Di chuyển tất cả nội dung từ các tab bên trái vào một trang duy nhất cho thiết bị đó.
> 
> Thật là nực cười khi phải đi xem liệu có sự kiện HMS nào không, hoặc có bản cập nhật chương trình cơ sở nào không.
> 
> Danh sách thanh bên của máy in sẽ hiển thị trạng thái đơn giản (như một phần của biểu tượng hoặc theo cách khác). Bằng cách đó bạn có một danh sách tất cả các máy in đã biết. Bạn có thể xem máy in nào đang ngoại tuyến. Bạn có thể xem máy in nào hiện đang in. Bạn có thể xem máy in nào đã in xong và cần xóa bản in để sẵn sàng.
> 
> Bạn có thể nói rằng tôi nghĩ tab thiết bị sẽ phát triển thành nơi quản lý máy in, xếp hàng in và tất cả các chức năng khác mà một người cần với nhiều máy in/một trang trại in.
> 
> Ngoài việc di chuyển thông tin từ các tab bên trái, ngăn trạng thái máy in cũng sẽ liệt kê cụ thể mọi bản in đã được xếp hàng đợi cho máy in đó. (Tôi giả định rằng việc quản lý máy in đầy đủ tính năng sẽ bao gồm việc xếp hàng các bản in theo cách mà máy in đầu tiên đáp ứng các yêu cầu cho bản in sẽ được sử dụng.

Vote

closed 2024-04-19 by github-actions[bot]