OrcaSlicer/OrcaSlicer #987: more info in file names

Gizmo6, 2023-05-07:
**Is your feature request related to a problem? Please describe.**
I switch a lot between materials and nozzle sizes depending on the product. I will sometimes forget which set up I have used.

**Describe the solution you'd like**
If the file name had some sort of info such as manual PA or auto PA, nozzle size used and material. I think it would be very helpful.

**Describe alternatives you've considered**
Manually putting this info in.

hliebscher, 2023-05-08:
you can add more variable in filename....
samples
https://wiki.bambulab.com/en/software/bambu-studio/placeholder-list

https://help.prusa3d.com/article/list-of-placeholders_205643. (names a little bit different as sample)

{input_filename_base}_{layer_height}mm_J1_{print_time}{day}{print_preset}{print_preset}.gcode

AK - a_0.25mm_J1_31m19s80.20mm 0.6 Standard @Snapmaker J1 - Snapmaker J1 0.6 Düse 1.6.3.gcode

julie777, 2023-05-08:
Do you mean the filename for the file that is sent to the printer?

Aron3DP, 2023-05-14:
I'd like to add to this request. I'd like at least filament temp to be added to naming. I've tried the tooltip items for nozzle_temperature and initial layer and they do not work.

Gizmo6, 2023-05-16:
> @Gizmo6 Do you mean the filename for the file that is sent to the printer?

Yes. so for example file output would be. "name of file" "nozzle size" "material type" so like. "Sample 0.8 PLA"

JamesKR3D, 2023-06-30:
I'd also like to have the print temperature variable to be added to the filename options.

VidiotGeek, 2023-09-03:
> I'd like to add to this request. I'd like at least filament temp to be added to naming. I've tried the tooltip items for nozzle_temperature and initial layer and they do not work.

I came here looking to see if I was missing something or if there really wasn't a way of pulling the temperature into the filename output. Seems like I'm note the only one. I've tried all the tool-tip names also, in {curly} and in [square] brackets. No dice.

solidus0079, 2023-09-10:
Looks like this is a thing now.  Not against the idea, I can certainly see the use of it - but is there an option to turn this off?  Or at least the print time suffix?  This change has added extra steps for me as I make tweaks to slices as I finalize settings I like for a model.  Before, saving _model.gcode_ into the same folder would overwrite the previous version I don't want anymore.

But now my first slice will give _model_PLA_1h19m.gcode_.  After printing I decide to try a different way and the second slice is _model_PLA_1h43m.gcode_.  After a few attempts I have a whole folder of files to delete, whereas before I did not as I would be saving the files over the old unwanted one.

Obviously, the save dialog allows right clicking and deleting, but the way Windows works it then changes the file name (including time) of the new one to the old one that now has an incorrect time.

If the option to turn this off is somewhere I did not see, I apologize.  Guidance requested.  :)

ding-bat, 2023-11-13:
someone please give a working example of temperatures in filename, thanks in advance

kylek29, 2023-11-16:
> > I'd like to add to this request. I'd like at least filament temp to be added to naming. I've tried the tooltip items for nozzle_temperature and initial layer and they do not work.
> 
> I came here looking to see if I was missing something or if there really wasn't a way of pulling the temperature into the filename output. Seems like I'm not the only one. I've tried all the tool-tip names also, in {curly} and in [square] brackets. No dice.

> I'd also like to have the print temperature variable to be added to the filename options.

> someone please give a working example of temperatures in filename, thanks in advance

nozzle_temperature is an array, so you have to reference a nozzle. [0] would be the first one.

`{input_filename_base}_{filament_type[0]}_{print_time}_{nozzle_temperature[0]}.gcode`

Outputs as:

`OrcaPlug_PLA_58m38s_230.gcode`

xi2023, 2023-11-20:
thanks @kylek29 
the only thing i was missig was a name/printer model in the filename.
printer settings -> notes -> type in a name or model
and add {printer_notes}

{input_filename_base}_{print_time}_{layer_height}mm_{filament_type[0]}{nozzle_temperature[0]}C_{printer_notes}.gcode

**filename_25m59s_0.2mm_PLA200C_printername.gcode**

kylek29, 2023-11-21:
> thanks @kylek29 the only thing i was missig was a name/printer model in the filename. printer settings -> notes -> type in a name or model and add {printer_notes} [...]

Yeah, that's one way to do it (just remember to not make your notes too big!). PrusaSlicer has assignable variables in that same area of the program (so you could assign a printer_name variable and reference it), but that feature hasn't been ported yet.

Echo-o-Tree, 2023-11-26:
> thanks @kylek29 the only thing i was missig was a name/printer model in the filename. printer settings -> notes -> type in a name or model and add {printer_notes} [...]

Just to add on to this, in case others like to sort by Material color/type you can go to Filament Settings > Notes and type the Brand/Color/Material etc there.  Then, in Process > Others > Filename Format use `{filament_notes[0]}` for single material prints, or `{filament_notes[0]}_{filament_notes[1]}` (add one per filament used for multi material prints)

This will append the filament notes per filament to the location you placed the line in your Filename Format section.

gidgiddoni2, 2024-02-29:
I'm a NB at Orca Slicer 1.9.1and would like to change the name of the .gcode fine to be plate_number print_time.gcode... Where in Orca do I set this filename format?

Catriks, 2024-05-23:
Is there a code for using current/project file name? Lets say I open a file called file123.stl, then save it as a doorhandle.3mf, and I want the gcode filename to be doorhandle.gcode. Using {input_filename_base} will still use the original filename.

jbjhjm, 2024-07-16:
@Catriks your problem is discussed here: https://github.com/SoftFever/OrcaSlicer/issues/1945

KubaMiszcz, 2024-08-06:
it doesnt work for me, I've tried your patterns and it always send `filename_plate_1`

where filename is name of 3mf file on disk, plate has no names, it update plate names if named, but nothing more, changing pattern wont affect generated gcode name

KubaMiszcz, 2024-11-06:
where should i set it? i'v changed field in: 
Process>others>filename format into "{input_filename_base}.gcode"
and nothing changed when i click print, same name in popup

Uranoman, 2024-11-16:
try this option:
{year}.{month}.{day}_{input_filename_base}_{filament_type[initial_tool]}_T{nozzle_temperature[initial_tool]}n{textured_plate_temp[initial_tool]}b_{print_time}_{total_weight}g{used_filament}m.gcode

lordestiny83, 2025-01-31:
I'm using this:

{print_time}_Support={enable_support}_Infill={sparse_infill_density}%_T={nozzle_temperature[0]}_B={hot_plate_temp[0]}_{layer_height}mm_{input_filename_base}.gcode

**Output:** 40m30s_Support=true_Infill=15%_T=240_B=85_0.2mm_BestCalibrationCube20mm_V1.gcode

KubaMiszcz, 2025-02-01:
it works only for exported .gcode, I want to use it to set name of file sending to printer when i click 'print'

OrangGeeGee, 2025-03-05:
> it works only for exported .gcode, I want to use it to set name of file sending to printer when i click 'print'

What I've found and wasn't obvious to me at first, is if there's any info in Project tab, it will use that model name when sending to printer. So if you've downloaded a .3mf file with metadata, printing ANY plate will use the same filename and get overridden in printer's sd card.

If I remove that info, reopen the file and Project section shows "no model information", only then the filename settings are respected.

Combining with info from [this issue](https://github.com/SoftFever/OrcaSlicer/issues/2752#issuecomment-1819799730), I've set my filename pattern to this and it different plates end up in the printer as separate files.
```
{input_filename_base}{if plate_name!=""}_{plate_name}{endif}_{filament_type[0]}_{print_time}.gcode
```

closed 2025-06-11 by github-actions[bot]