OrcaSlicer/OrcaSlicer #1005: QIDI X-Smart 3 wifi is not connecting

pbeeryRootTranslation, 2023-05-08:
**Describe the bug**
I've added a new QIDI X-Smart 3 printer. when I configure it for wifi I'm entering in the IP address similar to 192.168.68.74 and it doesn't connect. Instead, I get a dialog box (Orca Slicer error) saying that it could not connect to OctoPrint: curl:Couldn't connect to server: Failed to connect to 192.168.68.74 port 80: Connection refused [Error 7]       Note:OctoPrint version at least 1.1.0 is required.

This is a Klipper based machine from the factory. As far as I know OctoPrint is not installed on the QIDI X-Smart 3 printer. Other than not connecting to the machine remotely, the QIDI X-Smart 3 machine config is working well. Prints better than the default QIDIPrint software.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

fr3ak2402, 2023-05-08:
@pbeeryRootTranslation

I had created the profile for the X-Smart3. Thanks for your feedback.

Yes the Qidi X-Smart 3 comes from the factory with Klipper software, only the problem is unfortunately with the Qidi is that you used an interface than that of the Klipper.

Can you connect to the printer via QidiPrint?

pbeeryRootTranslation, 2023-05-08:
Yes, I can connect via QIDIPrint.  The IP address is correct. I don't see what port QIDIPrint is using but I've tried 80 and 443. I'm not really sure how to determine what port should be used.

fr3ak2402, 2023-05-08:
@pbeeryRootTranslation 

So does the 7125 port work for a connection with a Qidi X-Smart 3?

If so can you set this bug report to resolved?

pbeeryRootTranslation, 2023-05-09:
port 10088  gives the control panel with all the good stuff.  Port 7125 gives a connection status page.  I'm currently using 10088 and Orca is working perfectly. Thanks so much for you help.

closed 2023-05-09 by pbeeryRootTranslation

hidezosun, 2023-11-01:
I'm using QIDI X-MAX3.
I'm using orca v1.8.0, where can I specify the port number when connecting to the printer?

MarkusKgit, 2023-11-07:
@hidezosun 
> I'm using QIDI X-MAX3. I'm using orca v1.8.0, where can I specify the port number when connecting to the printer?

In the printer connection settings under hostname/IP/URL just enter the ip followed by a colon and the port, e.g.: `192.168.1.10:10088`

patones1, 2024-01-15:
Hello,
I had same issue when connecting to the printer from my Orca; but the printer is  the Anycubic Kobra 2 Max. Same error 7 (Failed to connect to 192.168.201.141 port 80)
I am not able to find the wifi communication port to connect to the printer from my Orca.
If somebody know I will appreciate 
Thanks

MarkusKgit, 2024-01-16:
> Anycubic Kobra 2 Max

Are you using the stock firmware on your Kobra 2 Max or did you flash Klipper? I don't think it will work with the Anycubic firmware.

patones1, 2024-01-17:
Thanks MarkusKgit,
I am using the original Kobra 2 Max firmware. 
I think I will have to flash Klipper. 
Have a nice day
Miguel