OrcaSlicer/OrcaSlicer #997: UI Slow to respond when changing settings tabs

noobydp, 2023-05-08:
**Describe the bug**
There is about a 1 second delay when clicking the settings tabs (Quality, Strength, Speed, Support, Other), and possibly other slow performance in the UI, but this is the most noticable.
It may not sound like much but it increases frustrating when changing a lot of settings.

Attached screen capture shows when the click event occurred so you can see the time between click and UI load.

I'm on Windows 11, Threadripper 2950X, Nvidia 2080Ti, 32gb ram.
Fresh/new install of Orca 1.6.2

**3mf File for This Bug**
N/A

**To Reproduce**
Open the UI, click the tabs

**Expected behavior**
The tabs should change instantly.

**Screenshots**
https://user-images.githubusercontent.com/5045555/236732613-bceee3e0-1e84-46b0-895e-60ea7b45019f.mp4
Video attached

**Printer model**
Custom voron trident, running a duet3 6hc with klipper

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 22H2 22621.1555
 - Orca Slicer 1.6.2

[image]

hliebscher, 2023-05-08:
not in my systems it woks fast...
MAC and WIN10  1.6.2/1.6.3

gaifeng8864, 2023-05-08:
> **Describe the bug** There is about a 1 second delay when clicking the settings tabs (Quality, Strength, Speed, Support, Other), and possibly other slow performance in the UI, but this is the most [...]

I have the same problem. Especially after slicing complex models, the preview interface is very slow and freezes.

mikebcbc, 2023-05-08:
This happens to me as well.

hliebscher, 2023-05-09:
only at all on OS WIN 11 ?

mikebcbc, 2023-05-09:
> only at all on OS WIN 11 ?

I'm on windows 10 64 bit

gaifeng8864, 2023-05-09:
> > only at all on OS WIN 11 ?
> 
> I'm on windows 10 64 bit

Me too

sashi, 2023-05-15:
Also same issue with Win 11
Ryzen 5600x
3060ti

quantumfrost, 2023-06-14:
This happens to me too, on Win 10. Switching tabs takes almost an entire second. Scrolling is slow too.

julie777, 2023-06-19:
There is an issue with how Bambu Studio/Orca work with graphics cards.
You might try settings/graphics and change Orca to use the energy saving setting. 
I gave details in a previous issue that you can search for using "graphics"

quantumfrost, 2023-06-19:
> There is an issue with how Bambu Studio/Orca work with graphics cards. You might try settings/graphics and change Orca to use the energy saving setting. I gave details in a previous issue that you can search for using "graphics"

My desktop only has an Nvidia GPU so I don't have the energy saving setting which makes it run on the integrated Intel GPU.

noobydp, 2023-06-20:
I also only have 1 dedicated Nvidia GPU

Amiyah14, 2023-10-31:
Do you have G-Sync enabled for windowed applications? Try switching G-Sync to full screen - fixed it for me. Even scrolling is fast now.

quantumfrost, 2023-11-07:
> Do you have G-Sync enabled for windowed applications? Try switching G-Sync to full screen - fixed it for me. Even scrolling is fast now.

I have G-sync turned off. Tried turning it on and setting it to full-screen just in case, still didn't help.

CrazyCoder, 2023-11-23:
Same issue. Scrolling inside the tabs is painfully slow too. Windows 10, RTX 3080, 5950X. GPU is set to performance mode in Nvidia Control Panel | Manage 3D settings.

160Hz screen refresh.

It also affects Bamboo Studio 1.7.7.89.

sandro-byte32, 2023-12-17:
Same for me scrolling is very slow.
Win 11 
Nvidia 2070 Super

DoomyD, 2024-01-20:
Same here, Orca 1.9.0
There is an obvious delay between navigating tabs, There is no way my PC ain't able to run handle it, it must be a software problem!

camled, 2024-01-21:
Orca 1.9.0, Windows 11 Pro
Scrolling through tabs is terribly slow. I turn the mouse wheel slightly and I have to wait a few seconds for it to scroll.

ellioteb, 2024-01-21:
Same for me too, issue when scrolling 

https://github.com/SoftFever/OrcaSlicer/assets/16530502/f3e771b6-a541-45bd-8b32-a4e2a0e2a210

### Computer Information:
Manufacturer: Gigabyte Technology Co., Ltd.
Model: X570 AORUS ELITE
### Processor Information:
CPU Vendor: AuthenticAMD
CPU Brand: AMD Ryzen 7 3700X 8-Core Processor
CPU Family: 0x17
CPU Model: 0x71
CPU Stepping: 0x0
CPU Type: 0x0
Speed: 3593 MHz
16 logical processors
8 physical processors
Hyper-threading: Supported
FCMOV: Supported
SSE2: Supported
SSE3: Supported
SSSE3: Supported
SSE4a: Supported
SSE41: Supported
SSE42: Supported
AES: Supported
AVX: Supported
AVX2: Supported
AVX512F: Unsupported
AVX512PF: Unsupported
AVX512ER: Unsupported
AVX512CD: Unsupported
AVX512VNNI: Unsupported
SHA: Supported
CMPXCHG16B: Supported
LAHF/SAHF: Supported
PrefetchW: Unsupported
### Operating System Version:
Windows 11 (64 bit)
NTFS: Supported
Crypto Provider Codes: Supported 311 0x0 0x0 0x0
### Video Card:
Driver: NVIDIA GeForce RTX 3060
DirectX Driver Name: nvldumd.dll
Driver Version: 31.0.15.3742
DirectX Driver Version: 31.0.15.3742
Driver Date: 9 12 2023
OpenGL Version: 4.6
Desktop Color Depth: 32 bits per pixel
Monitor Refresh Rate: 59 Hz
DirectX Card: NVIDIA GeForce RTX 3060
VendorID: 0x10de
DeviceID: 0x2504
Revision: 0xa1
Number of Monitors: 2
Number of Logical Video Cards: 2
No SLI or Crossfire Detected
Primary Display Resolution: 3840 x 2160
Desktop Resolution: 5760 x 2160
Primary Display Size: 27.56" x 15.35" (31.54" diag), 70.0cm x 39.0cm (80.1cm diag)
Primary Bus: PCI Express 16x
Primary VRAM: 12287 MB
Supported MSAA Modes: 2x 4x 8x
### Memory:
RAM: 49072 Mb

pawg, 2024-01-21:
Orca 1.9.0, Windows 11 Pro here also.
7950x3D, RAM 64Gb, RTX 3090.

At that point it's clear that it's not hardware related. However it clearly prevents me from using this brilliant piece of software as a daily driver. Hope it could be resolved soon because the rest is lovely!

ElvisTheKing, 2024-01-29:
Orca 1.9.0, Windows 11 Pro
Intel 10900 RTX2080
Tabs are slow both on intel and nvidia GPUs

lixxbox, 2024-02-02:
I have the same problem. Switching tabs takes very long and scrolling is stuttering.
Ryzen 3700x, 32GB RAM, Windows 11, OrcaSlicer 1.9.0

However, I got the Linux version running on Windows with WSL. It is just as snappy as you would expect...

h3rm, 2024-02-03:
i have the same problem on WQHD-resolution (2560px) in full screen (2 wqhd screens/multimonitor) and 144hz.. resizing the window helps a bit

scruffyOrc, 2024-02-14:
I have this same issue, 2x Xeon x5690, 96GB RAM, Radeon RX 6600, WIN 11, Orca Slicer 1,9,0 (and an old 1.7.0 and a 2.0.0 beta from FlashForge Official's fork too all have the issue so it's been around for a while and it's not going away).

barnowan, 2024-02-18:
Same issue here with slow tabs and scrolling on my laptop. My desktop works just fine.

szafranski, 2024-02-28:
Same here, only on Windows 11. MacBook Air - works fine.

wilkems, 2024-03-04:
Same thing here. Windows 11, Ryzen 7 3700x, Nvidia GeForce 2070 Super. Bambu and Prusa are working fine. 

Based on the comments, is everyone on Nvidia?. I'm seeing both Ryzen and Intel, though mostly Ryzen.

szafranski, 2024-03-04:
I discovered that saving and reopening same project sometimes fixes the issue. Weird.

RobBott, 2024-03-11:
Same issue for me on Windows 10 and an Nvidia GPU

LeHungryBoi, 2024-03-16:
same issue with tab switching and scrolling, ryzen 1600 and rx 580

vgdh, 2024-03-16:
same issue
rtx460
Intel(R) Xeon E31270

i8086m, 2024-03-24:
Same issue
Windows 11
i7-8086K
RTX 3070
32GB RAM

alexyubin, 2024-03-27:
Same issue

Akucuki, 2024-04-12:
I'm on Arch Linux, launching it using appimage and it's painfully slow, device tab sometimes doesn't work at all.

Akucuki, 2024-04-12:
The problem on Linux seems to be with the webkit2gtk with versions higher than 2.42.5-2. Downgrading back to 2.24.5-2 eliminates all of the issues with blank and laggy tabs. Works like charm now.

Tethgar, 2024-04-22:
Issue still persists on v2.0.0, scrolling works fine when the application is first opened but eventually scrolling becomes very laggy, the rest of the application works just fine and is responsive. Restarting Orca fixes the issue but obviously don't want to be doing this every 1-2 hours

Flavourdynamics, 2024-04-25:
I'm also experiencing this issue, gradual slowdown.

epicylon, 2024-05-09:
Same issue. NVIDIA 2080, Intel i7 CPU. 64GB RAM.

wilkems, 2024-05-10:
I just doubled my ram from 32 to 64gb, and I upped the speed from 2133, to 3200mhz and it made a massive improvement. 

Not sure if that's brute forcing over the actual issue, or there's a bug with memory allocation that my upgrade mitigated, but wanted to report back the change.

noobydp, 2024-05-10:
I've had extended periods (months) where it works fine, to the point I forgot I raised this ticket. 

Aaaaaand then it happens again and I remember the pain. 

I don't want to discount your experience, but I'm also on powerful hardware, as are many.

szafranski, 2024-05-10:
At least works fine on Mac. (I know it is not suffcient.) I think it is something related to memory assignment etc on PC but I am not an expert in this regard.

wilkems, 2024-05-10:
I'm unfamiliar with the structure, but is there a point where Orca devs see this and step in to look at it? Or is it up to the community to narrow it down or wait until a working version is released?

pluciorx, 2024-05-16:
Could be the same issue as: #3899

wilkems, 2024-05-16:
> I have found one problem and have a fix on my machine that is looking solid so far. Changing tabs is almost instant now. I will probably PR it and see if it gets in.
> 
> The whole UI is really badly written and severely unoptimised so there are likely many places where this could be sped up significantly.

What were you able to try that's seeming to work? I monitored ProcMon and on tab change, about a million image files were called, making me think it's a storage access issue.

akoolenbourke, 2024-05-17:
> What were you able to try that's seeming to work? I monitored ProcMon and on tab change, about a million image files were called, making me think it's a storage access issue.

What I did is avoid any subsequent rebuilds of the UI when a tab has already been visited. The current code essentially throws away all the contents of the tab you're leaving and rebuilds the one you're entering. Then if you go back to the previous tab it does the same. My fix simple hides the contents of tab pages when you leave the page so it doesn't rebuild it from scratch.

This is my first time in the code base and I'm relatively new to the software so there could be something I'm missing but it's so much better now.

I would guess that rebuilding is the thing that does a tonne of image work, likely for icons and things and I imagine there are MANY places where optimisations can be done.

Here's a video of what it's like for me now, compared to about 1 second whenever I switched to the Quality tab. I will do a PR soon and see if it gets in.

https://github.com/SoftFever/OrcaSlicer/assets/29034122/1678b022-2d1e-41c7-b1b6-baa1bc00fd17

wilkems, 2024-05-17:
> > > I have found one problem and have a fix on my machine that is looking solid so far. Changing tabs is almost instant now. I will probably PR it and see if it gets in.
> > > The whole UI is really badly written and severely unoptimised so there are likely many places where this could be sped up significantly.
> > 
> > 
> > What were you able to try that's seeming to work? I monitored ProcMon and on tab change, about a million image files were called, making me think it's a storage access issue.
> 
> What I did is avoid any subsequent rebuilds of the UI when a tab has already been visited. The current code essentially throws away all the contents of the tab you're leaving and rebuilds the one you're entering. Then if you go back to the previous tab it does the same. My fix simple hides the contents of tab pages when you leave the page so it doesn't rebuild it from scratch.
> 
> This is my first time in the code base and I'm relatively new to the software so there could be something I'm missing but it's so much better now.
> 
> I would guess that rebuilding is the thing that does a tonne of image work, likely for icons and things and I imagine there are MANY places where optimisations can be done.
> 
> Here's a video of what it's like for me now, compared to about 1 second whenever I switched to the Quality tab. I will do a PR soon and see if it gets in.
> 
>  Fastertabs.mp4

That checks out with what I was seeing in ProcMon. So much file access. Dont' know why it doesn't load just once and cache it for recall (talking out of my rear. I don't know how this program works). 

Did you also have the slow scroll issue? If so, did this fix it too?

akoolenbourke, 2024-05-17:
> That checks out with what I was seeing in ProcMon. So much file access. Dont' know why it doesn't load just once and cache it for recall (talking out of my rear. I don't know how this program [...]

After my change procmon reports zero file accesses when going to the tabs a subsequent time which is good. From a brief test it doesn't (And I wouldn't have expected it to) fix the slow scroll.

PR has been made so let's hope it's a good fix and makes it in.

wilkems, 2024-05-17:
> After my change procmon reports zero file accesses when going to the tabs a subsequent time which is good. From a brief test it doesn't (And I wouldn't have expected it to) fix the slow scroll.
> 
> PR has been made so let's hope it's a good fix and makes it in.

PR? Please excuse my lack of understanding of the vernacular. Is your build something that I would be able to try? And have you noticed if it helped the slowing down of the whole computer?

Thanks for your work!

akoolenbourke, 2024-05-18:
> PR? Please excuse my lack of understanding of the vernacular. Is your build something that I would be able to try? And have you noticed if it helped the slowing down of the whole computer?
>
> Thanks [...]

PR means Pull Request. It's what you make when you have a change you've done that you would like put into the main project. The maintainer of the project can then see if it's a good thing to go in and check it doesn't break anything etc.

Sascha8, 2024-06-25:
Any news to this topic? 
I have the same lag after a while working with the slicer. Just came here to create an issue after I took an hour to find some reproducible steps to describe it. But I see here are already many info about this issue. Reading about rebuilding/reloading much data fits to my observations - clicking and scrolling in the left pane is horrible slow (up to 5 Seconds on a powerful machine). Dears devs, please give this some more priority. At least you can give it some more notice and maybe ask for more details if it is hard to identify. Thx

L2on1, 2024-06-25:
Same issue here, my orcaslicer is very very slow and laggy.
And especially, I wait for responce (windows) when I want to save my project and when I want to flush filament or add new filaments.

ASUS zephyrus G14 : Windows 11 / AMD Ryzen 7 4800HS / rtx 2060

I've already tried solutions from #3899

aaronatweta, 2024-06-25:
> Any news to this topic? 
> 
> I have the same lag after a while working with the slicer. Just came here to create an issue after I took an hour to find some reproducible steps to describe it. But I see here are already many info about this issue. Reading about rebuilding/reloading much data fits to my observations - clicking and scrolling in the left pane is horrible slow (up to 5 Seconds on a powerful machine). Dear devs, please give this some more priority. At least you can give it some more notice and maybe ask for more details if it is hard to identify. Thx

There was a fix for the navigation lag that me and someone else did and did a PR for but they reverted it for some reason.

ApexArray, 2024-06-25:
Also experiencing this, and it's incredibly frustrating. Are there any known workarounds at this point?

Windows 11
Ryzen 5 3600
24GB RAM
GTX 1080ti

ellioteb, 2024-06-30:
I have this issue too
Windows 11 Pro 23H2 22631.3593
Ryzen 7 3700X 8-Core Processor   
48.0 GB
RTX 3060 12gb

Sleff1, 2024-07-24:
Same here 
windows 11 pro
intel i5 2400
gtx 750ti
12gb of ram 
ssd

wilkems, 2024-07-28:
Is anyone experiencing this on a single 1080p monitor? I resized the Orcaslicer window and instantly a huge improvement. I'm running 3 2560x1440 and one 2560x1080. Trying to find the common thread, and the straw I'm grasping at this time is total resolution vs graphics horsepower.

Adib-C, 2024-07-29:
> Is anyone experiencing this on a single 1080p monitor? I resized the Orcaslicer window and instantly a huge improvement. I'm running 3 2560x1440 and one 2560x1080. Trying to find the common thread, and the straw I'm grasping at this time is total resolution vs graphics horsepower.

Yes i'm experiencing this on a single 1080p monitor

wilkems, 2024-07-29:
> Yes i'm experiencing this on a single 1080p monitor

Well darn 😂. What's your graphics card setup?

popos123, 2024-07-29:
The GPu doesn't matter, the same is on Intel, and on the 3090 on FHD and on 4K I was tested on multiple configurations, the BUG is serious

wilkems, 2024-07-29:
It's driving me insane and killing my work flow. What does it take to get devs involved?

Adib-C, 2024-07-29:
> Well darn 😂. What's your graphics card setup?

AMD RX 6650 XT

[12 more replies]