OrcaSlicer/OrcaSlicer #998: Slow down inner perimeters on overhangs, not only outer perimeters

rovili, 2023-05-08:
I noticed that only the outer perimeters are slowed down on overhangs, (inner) perimeters are not affected at all.
This is causing curling on steep overhangs, especially above 45°. 

Please give us the option, to also slow down the perimeters. (such as PS and SS does)

rovili, 2023-05-23:
I would like to push this, as it is crucial for fast printing for object with steep overhangs.

renemosner, 2023-06-21:
I would also ask for this option in PrusaSlicer also slows down on internal walls and it works great see pictures.  Uncooled or deformed overhanging walls are then not formed.
[image]

scottbmc, 2023-08-10:
*push*

Would really appreciate this feature since i have a hard time on overhangs especially with ABS. So i think this could really improve overhang quality.

igiannakas, 2023-09-18:
This is now implemented as part of the curled edges detection feature introduced in the latest nightly. It will reduce internal speed when there is a risk of the perimeter curling up.

please close as complete.

https://github.com/SoftFever/OrcaSlicer/pull/2056

closed 2023-12-26 by github-actions[bot]

FlowForm, 2024-01-10:
I can't see that this new function is working in my situation. - Would it be possible to set a certain angle, where it would slow down on internal walls? @

[image]

igiannakas, 2024-11-25:
> I also can see no difference with the curled perimeters parameter turned on or off. The speed stays the same, and the curling still exists

Try slicing a standard benchy and look at the tip of the hull. Post some pics of your overhang speed settings too please

Sjullep, 2024-11-25:
Left is the benchy, and the curled perimeters parameter is doing it's job. But in the right object (a rubber duck, oval shape) a cannot see any difference for the inner walls when i turn the parameter on or off. 
I have problems with the wall curling up when printing the rubber duck, so i hoped that setting the curled perimeters parameter to On would do the trick. Bur it looks like it only works with sharp corners?

Curled Perimeters = ON
[image]

Curled Perimeters = OFF
[image]

igiannakas, 2024-11-26:
This feature is more prominent on sharp corners as they are the ones that tend to lift up.

for the round objects just adjust your overhang speeds as these should be covered by them.

Sjullep, 2024-11-26:
Even when i set all the overhang speeds to 1, the speed of the inner walls doesn’t change. Looks like they are not recognized as overhangs, just the outer perimeter is

igiannakas, 2024-11-26:
The inner wall speed won’t change as they are not overhangs - they have a full overlap with the perimeters below. And as they are not on sharp edges (or on extreme angles) the curled perimeter detector doesn’t affect them. So they won’t slow down.

if your internal perimeters are curling up with this model, more likely than not you’re either printing too hot or printing too fast for the melt capacity of the nozzle.

Sjullep, 2024-11-26:
_> The inner wall speed won’t change as they are not overhangs - they have a full overlap with the perimeters below. And as they are not on sharp edges (or on extreme angles) the curled perimeter detector doesn’t affect them. So they won’t slow down._
I had a feeling this was the case... The overhang angle is between 75 and 60%, so rather steep. But still the inner wall speed not affected bij the parameter. 
A smaller overhang part is printed ok (like the beak of the duck), the curling just occurs when there is a overhang with a large perimeter, like the lower part body of the duck. 

_> btw on the picture above you do see the curled perimeters algorithm slowing down your lower layers considerably._
With the benchy, it does slow down. But with the duck, it doesn't, just the outer perimeter. The inner perimeters are printed with a speed of 184, regardless the setting of the curled overhang perimeters. Only in the first few layers a see a difference in speed in the inner perimeters, and only in the relatively sharp corner in the front of the duck. So i guess this is how it's suppose to work?

I will try to lower the temperature. Lowering the speed would effect the print time, while a lower speed is only necessary for the lower part of the duck where the overhang with the large perimeterts is, so i would like to avoid that.
I wil post my findings later. Thanks so far.

theyyg, 2025-08-15:
I have a printer that is also struggling with this.  Printing any cooler compromises the layer adhesion, and the system prints other areas without problems.  This problem is specific to overhangs.  This suggestion seems like it would address the issue, but the implementation doesn't.

It appears that I have a misunderstanding of "overhangs" as referenced in this chat, so I want to be clear that I consider the entire wall (both inner and outer loops) that has an angle difference from the upward direction to be an overhang, when a higher layer would extend beyond a lower layer.  My definition is broader than what I've seen referenced here.  It applies not only to the loop/wall that is printed without full support from a layer underneath but also to the nearby walls.  

I suggest that a wall that printed on top of a true overhang also has need for extra cooling time.  This attribution could reasonably extend to walls on the next higher layer, with a falloff to prevent it from extending forever.  

Since corner detection is required in order to compensate for curling, the "slow down for curled perimeter" option is not sufficient in its current state.  The slowdown for the outer walls needs to extend at least partially to the inner walls.

> if your internal perimeters are curling up with this model, more likely than not you’re either printing too hot or printing too fast for the melt capacity of the nozzle. This one should be an easy model to print.

Yes, I agree that the cooling capabilities are not sufficient to print the entire wall of an overhang area at these speeds without curling.  This is the exact reason why we're asking for this feature.  There are many areas of the model that print with an acceptable quality at full speed everywhere except where the model has an overhang angle.  Implementing this feature will allow faster printing because it allows us to slow down only the part of the model that needs the additional cooling time rather than slowing down the entire print by reducing the overall print speed.

I would appreciate if this feature could be revisited and extended to make it better for everyone.