OrcaSlicer/OrcaSlicer #1014: Extruder Clearance Input filed broken

tome9111991, 2023-05-09:
Hi,
after switch from english to german language the text input fied is broken from Extruder Clearance.
in german version slide down the field 2 lines.
[image]

after switch back to english vesion  slide down the field 1 lines.
[image]

Radius should be 65mm
Heigh to Rot = 36mm

Printer Voron 2.3 300

Issue only on 1.6.3 beta i dont know what build i use.

tome9111991, 2023-05-09:
i tried this build https://github.com/SoftFever/OrcaSlicer/actions/runs/4925890271
still same issue

hliebscher, 2023-05-09:
is fixed with #1012 
[image]

tome9111991, 2023-05-10:
sorry i tried some build but still issue.
i can change language from german to english still have this bug

hliebscher, 2023-05-12:
i can change any language

tome9111991, 2023-05-22:
i have this issue still in 1.6.3 pre-release

SoftFever, 2023-05-22:
Maybe a Graphic Driver issue.
Have you tried to update the GPU driver?

tome9111991, 2023-05-23:
Yes i have tried latest driver and default driver from windows10

closed 2023-11-21 by github-actions[bot]