OrcaSlicer/OrcaSlicer #1015: Only Z hop for a range of layers

gbyleveldt, 2023-05-09:
Is it possible to make z hops only for a range of layers like in Superslicer? It’s useful when doing multicolour prints using a single nozzle with manual filament changes for different objects occupying the same space

Basically, it’ll only z-hop from layer 1 to 3 (for example) so it doesn’t crash into previous print

hliebscher, 2023-05-09:
You can add a second filament with z Hop enabled in filament overwrite Settings and change this for your Layers that you need

gbyleveldt, 2023-05-10:
Thanks for your advice. I’m trying to think how I can make that work for my application but coming up empty. I’m aware of the filament overrides but making it work (if I can) would be clunky. Maybe I’m not understanding your suggestion properly?

blastrock, 2023-05-20:
Prusa Slicer allows enabling Z-hop only on a range of height (Only lift Z "above Z" and "below Z"). Another use I have for that is to disable z-hop on first layer as it causes adhesion issues with some filaments for me.

closed 2023-11-21 by github-actions[bot]