OrcaSlicer/OrcaSlicer #1022: Custom Tool Change Gcode

Xcorpia07, 2023-05-09:
The issue is that there is a filament change with M600 but the tool change is made after the movement prior to the purge tower, which makes it impossible to easily modify this area of ​​the gcode.

I wish Orcaslicer could bring over the multimaterial features of PrusaSlicer or SuperSlicer. Since in my printer without AMS, but with ERFC I need a specific gcode for material changes and purges.

Like me there are many multimaterial ERFC users who would be very grateful for this function.

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

[image]

[image]

hliebscher, 2023-05-09:
Witch Code do you need?
Can you Place at filament End Gcode or filament Start Gcode 
It was executed every change

Xcorpia07, 2023-05-09:
I need is to change the output of the tool change. so that it runs some previous macros without having to modify the whole system.

 I've also seen that Orcaslicer doesn't ram the purge tower. Nor any adjustment of tips. Since bambu lab uses a cutter on extruder.

PatrikTheDev, 2023-06-05:
How did you add multiple extruders to your printer model in Orca? As far as I know it’s impossible to create a custom multimaterial printer (multi material with a single nozzle) because the menu isn’t present (at least on my machine)

Xcorpia07, 2023-06-05:
Just by adding new filaments in the filaments section. You can add the same material and then change the color, the operation is the same as Bambu Studio, you can see the manual for it.

[image]

Chaos-Seed, 2023-06-08:
Agreed @Xcorpia07, This would make Orca Slicer the go-to for me and a lot of others using ERCF. Thank you for raising the Feature Request!

mojotaker, 2023-09-03:
Yes please add multi material support for ercf. Thank you.

closed 2023-12-10 by github-actions[bot]