OrcaSlicer/OrcaSlicer #1023: Bridge fan speed not working 1.6.2

TechDude23, 2023-05-09:
The bridge fan speed is not working it will show the fan @ 90 in the preview screen but on the Bambu lab x1 display it will not go to 90 and is effecting parts. 

Windows 10
Orca V 1.6.2  

[image]

SoftFever, 2023-05-09:
Did you check the generated G-code to see if the M106 command is included before printing the overhangs?

TechDude23, 2023-05-10:
G1 F3253
G2 X108.581 Y119.186 I-10.105 J-77.234 E.38956
G3 X113.057 Y117.629 I9.863 J21.136 E.18982
G1 X113.221 Y117.591 E.00673
G1 F900
M106 S229.5
G1 X113.241 Y117.587 E.00082
G1 X113.241 Y134.381 E.67144
G1 X113.221 Y134.376 E.00083
M106 S150.45

I do see the M106 in the G-code but even if I run a hangover test to get the lower overhang layers to look good I have to set the max fan speed threshold to 90 then the 45 overhangs will look good I know in the past I would see the fan % change on the Bambu LCD display unless the newest Bambu Lab X1 firmware 1.5.01 has changed something or V1.6.2 and the sound of the parts fan seams to stay the same db. level also.

SoftFever, 2023-05-10:
It's likely that the issue you're experiencing is related to the fan kick-in time. 
Fortunately, @scottmudge has recently ported a fan speed delay feature from SuperSlicer into OrcaSlicer. You can find more information about it in PR #910. 
This feature should help to improve the issue you're facing. 
The latest nightly build of OrcaSlicer includes this change, so you can give it a try and see if it helps.
[image]

TechDude23, 2023-05-10:
Thank you SoftFever I will test it out how do I get this release to test ?

scottmudge, 2023-05-10:
This is the latest nightly:

https://github.com/SoftFever/OrcaSlicer/actions/runs/4925890271

TechDude23, 2023-05-10:
Thank you I downloaded it and I will update you

SoftFever, 2023-05-10:
> G1 F3253 G2 X108.581 Y119.186 I-10.105 J-77.234 E.38956 G3 X113.057 Y117.629 I9.863 J21.136 E.18982 G1 X113.221 Y117.591 E.00673 G1 F900 M106 S229.5 G1 X113.241 Y117.587 E.00082 G1 X113.241 Y134.381 E.67144 G1 X113.221 Y134.376 E.00083 M106 S150.45
> 
> I do see the M106 in the G-code but even if I run a hangover test to get the lower overhang layers to look good I have to set the max fan speed threshold to 90 then the 45 overhangs will look good I know in the past I would see the fan % change on the Bambu LCD display unless the newest Bambu Lab X1 firmware 1.5.0.1 has changed something or V1.6.2 and the sound of the parts fan seams to stay the same db. level also.
> 
> I closed down Orca slicer and I started running a new test again I can see the first top layer bridge speed go to 90% fan speed on the Bambu lab LCD display.
> 
> Yep the overhang at 30% is not kicking fan up to 90%

BTW, it should be a firmware bug in Bambulab if the fan does not respond to the M106 command properly.

TechDude23, 2023-05-10:
Ok I ran some more 45 and 55 degree overhang test and the fan on the newer version 1.6.3 will throttle the fan 60% - 70% at times on overhangs and will show on the LCD display and you can hear the fan change speeds so I installed the new version of Bambu Lab slicer and it will do the same thing so then I used my ideaMaker slicer and the fan will run at 90% with a 30 degree or more overhang and not shut off or throttle it will also show on the Bambu Lab LCD display and you can hear the fan staying on @ 90% so the new firmware is ok with the M106 G-codes fan commands. 

I also tested the new fan time settings and it did not solve the overhang problem.    

The slicer needs an option to have the cooling overhang threshold fan to stay on and not throttle the part fan; it will not even go to 90% @ 55 degree unless it is a bridge. 

This might also be the reason for all the complaints with round holes in side walls but I have not looked into that yet but I do think the new fan time setting will help if the fan is at the correct overhang speed.

Great job on the new feature added :)  Thank you Team Ocra

Mario-vminion, 2023-05-10:
for the setting how many seconds dos it has to be.. now its 0

scottmudge, 2023-05-10:
> for the setting how many seconds dos it has to be.. now its 0

Basically:

* Set to `0` means it's disabled.
* Set to `Fan speed-up time` to any positive number to move the time at which the fan is turned on backward in time by that number of seconds. Can be fractional (e.g., 0.5 seconds). My fan takes ~0.5-0.7 seconds to get up to speed, so I set this to 0.6 seconds.
* Set 'Fan kick-start time' to any positive number to cause the fan to go to maximum speed for that number of seconds (can be fractional) to "kick-start" it up to speed a bit faster. I generally have this set to 0.1 seconds. Do note that this kick-start time is moved to the start of the `Fan speed-up time`, if that set to a non-zero value.

SoftFever, 2023-05-11:
> Ok I ran some more 45 and 55 degree overhang test and the fan on the newer version 1.6.3 will throttle the fan 60% - 70% at times on overhangs and will show on the LCD display and you can hear the [...]

The slicer tells the printer to speed up the fan at the right place/time, but the printer didn't respect the code instruction.
From the gcode and preview, the slicer told the printer to keep the fan at 90% speed.

Can you share the project? 
Also, the gcode generated from ideamaker

TechDude23, 2023-05-11:
Ya I think this should work so you can figure it out 👍 

[Ocra Bracket.zip](https://github.com/SoftFever/OrcaSlicer/files/11454888/Ocra.Bracket.zip)
[ideaMaker Bracket.zip](https://github.com/SoftFever/OrcaSlicer/files/11454889/ideaMaker.Bracket.zip)

shauneccles, 2023-05-12:
I note the G2 in your gcode

PR#910 states

        " It assumes infinite acceleration for this time estimation, and will only take into account G1 and G0 moves (arc fitting"

Could this be a confounder?

TechDude23, 2023-05-13:
> I note the G2 in your gcode
> 
> PR#910 states
> 
> " It assumes infinite acceleration for this time estimation, and will only take into account G1 and G0 moves (arc fitting is not supported)"
> 
> Could this be a confounder?

Hello shauneccles,

I don't know on the G2 but I know SoftFever team did optimized M106 command so I will have to test it out with the new beta release today 👍 

Thank you for looking at code :)

closed 2023-11-21 by github-actions[bot]