OrcaSlicer/OrcaSlicer #1025: Windows not rendering properly...

neogenix, 2023-05-10:
**Describe the bug**
When I load OrcaSlicer 1.6.2 it doesn't render window level text. 

**To Reproduce**
Run on OrcaSlicer

**Screenshots**
[image]

**Printer model**
X1 Carbon

**Desktop (please complete the following information):**
 - OS: Windows 10 running Atlas. 
 - Version: Atlas OS

SoftFever, 2023-05-10:
@neogenix 
You will need to install two dependencies.
Please refer to: https://github.com/SoftFever/OrcaSlicer#how-to-install

neogenix, 2023-05-13:
> @neogenix You will need to install two dependencies. Please refer to: https://github.com/SoftFever/OrcaSlicer#how-to-install

Update: 

Both were installed, but I repaired the Visual C++ item, and that fixed it. Thanks!!

closed 2023-05-13 by neogenix