OrcaSlicer/OrcaSlicer #1026: Supporting dense layer from superslicer

max-cz, 2023-05-10:
**Is your feature request related to a problem? Please describe.**
Solid layer cannot be supported with low infill

**Describe the solution you'd like**
Add supporting dense layer as on Superslicer

rfnovo, 2023-05-11:
On the same topic, the "Above the bridge" flow ratio is also a valuable option to improve top layer quality over infill, when using < 5 top layers.

rfnovo, 2023-05-11:
Just to add an example, here is the same model, printed with the same parameters (20% infill, 3 walls, 3 top, 3 bottom), only difference is that one was sliced in Super Slicer with dense layer and 120% above bridge and you can clearly see the difference of quality in the top layer. 
[image]
[image]

max-cz, 2023-05-11:
I'm not talking about bridges.

Even if I use 120% this(as example) will not be supported:
[image]

While in SS it create a "net" to support the solid infill:
[image]

This is under infill options, and it very useful with low infill.
[image]

rfnovo, 2023-05-11:
I know, but this setting also affect the top layer quality because as the name implies it add an extra support for the top (dense) layer. Maybe I should had create a separate post for the above bridge flow. Sorry if it seems I tried to hijack your post. 
[image]
[image]

max-cz, 2023-05-12:
No worries, its related to top solid layers, and could maybe be easy to implement together.

3Dxdream, 2023-05-24:
It is strange, that this "feature" is missing in such a nice slicer. Printing and changing path direction in midair, without any anchoring (infill or perimeter), is, to be honest, a slicing fail. I hope for a fix, otherwise OrcaSLicer is not usable, at least for me.
These bridge lines or first solid layer lines must be anchored to the underlying infill lines or perimeters (should be profile settings).
And there should never be a printed line (perimeters, bridges, infill, ....) which changes direction in the air. This could be devastating for the whole print!

max-cz, 2023-05-24:
> It is strange, that this "feature" is missing in such a nice slicer. Printing and changing path direction in midair, without any anchoring (infill or perimeter), is, to be honest, a slicing fail. I [...]

Superslicer is the only one I know that has this the way to "fix" it increase the infill density, but it come at the cost of more material use.

3Dxdream, 2023-05-26:
> Superslicer is the only one I know that has this. The way to "fix" it increase the infill density, but it come at the cost of more material use.

I am using Superslicer and Cura. In Cura it's in the setting :"Top/Bottom --- Skin Expand Distance", where you can fill in a larger number in [mm] to get a properly applied supporting layer (spans over the whole area between perimeters) for the first solid layer.
Increasing the infill density, is not only more material usage, higher print time, more weight,.... but also no correct fix. The dangling molten plastic lines are just shorter. They too melt and accumulate at the nozzle surface an cause unnecessary printing artifacts.

3Dxdream, 2023-05-30:
Finally, there is hope!
https://github.com/SoftFever/OrcaSlicer/issues/1165

DAKarp, 2023-07-15:
+1 This is the one lacking feature preventing me from switching from SuperSlicer to OrcaSlicer 🤞

mlewis-everley, 2023-07-31:
> 

Same here! I have tried OrcaSlicer and it is pretty awesome, but no dense infill and no brim ears (I know this is on the roadmap) means I cannot use it :-(

mlewis-everley, 2023-11-08:
So, the latest release of Orca has a variant of dense infill that I am testing now (but seems to work well). As far as I am concerned this issue is probably closed? What do you think @max-cz ?

max-cz, 2023-11-08:
> So, the latest release of Orca has a variant of dense infill that I am testing now (but seems to work well). As far as I am concerned this issue is probably closed? What do you think @max-cz ?

Yes, we can close it, Extend sparse infill seem pretty close to this.
Thanks

closed 2023-11-08 by max-cz

TheAnimatrix, 2024-09-10:
Superslicer's version is still superior. Orca gives me layer tearing even on the latest version like this
[image]

julian-weinert, 2025-12-25:
Support Dense Layers is far superior. It solves additional problems and improves top surface quality by a LOT.
@mlewis-everley would you please reopen this issue and consider implementing the feature?
I'm really struggling getting the same quality over large areas with OrcaSlicer, even when I add an additional top layer. It simply doesn't do nearly as well.